opentx/opentx #2680: Differential from a channel

mrabault, 2015-08-09:
Hello,

I run OpenTX 2.1.1 on my Taranis Plus. For my 3m glider, I've setup aileron differential, controlled by the GVAR GV1. However, I'd like to configure differential suppression. To do that, I set a curve up and a high mix multiplying crow by GV1. That works and gives me a channel that returns GV1 unaltered when crow is minimal and 0 when it is maximal. Now, to make aileron diff use this channel's value, I could create another GVAR that is the *effective* aileron differential.

However, this is a bit clunky IMHO. What would be better would be allowing differential to use fixed values, GVARS *or* output from a channel. I think that'd increase flexibility.

Thanks in advance for your attention and congrats for having developed such a great OS for TXs !

kilrah, 2015-08-09:
Now that GVARs are sources like others there's no reason why we couldn't have the entire source list available everywhere when in "variable" mode instead of just the GVARS, is there?

mrabault, 2015-08-09:
That'd be awesome ! Plus it makes sense: there's a bunch of inputs, let them be available from everywhere.

bsongis, 2016-11-16:
Not an easy one without a big EEPROM change: GVARS are just values that are not used in the range. Here we would have such a structure (from @mrabault email):

PACK(typedef struct {
    uint8_t type;
    union {
        int8_t numeric;
        mixsrc_t reference;
    }
}) value_t;

But this cannot fit into a simple 8bits value as we have today

closed 2020-07-08 by stale[bot]