opentx/opentx #2681: Test the X9E watchdog reset

projectkk2glider, 2015-08-09:
Have we tested what happens with the 9XE power when the watchdog reset happens? Does it survive the reboot and remains powered up? Does the value of additional capacitor from the RTC issue #2626 affect this in any way?

bsongis, 2015-08-09:
Yes I tested it on mine, but I don't have the capa, even no 100uF!

projectkk2glider, 2015-08-23:
Here is a special OpenTX build for 9XE that includes a test for a watchdog reset and instructions on how to use it:

**WARNING** this version of OpenTX is for **testing only**. **Do not fly with it**. Provided bin file is for Taranis 9XE only. I will not work on other Taranis radios!

First **make sure you have a backup of your current radio firmware**. You will need it to revert back to your existing setup after the test is done. You should also have a **backup of your current models and settings** (EEPROM) in case things go south.

Steps to test:
* make backups of your current firmware and settings
* flash the test firmware (link below)
* do the test (instructions below)
* report back the result of the test (here)
* flash back your backed-up firmware and if neeeded restore models and settings back.

The provided bin file is in English language and includes the default options. It is build from a pre-release version of 2.1.2 (branch https://github.com/opentx/opentx/tree/projectkk2glider/issue_2681_watchdog_test)

Watchdog test instructions:
* start the radio normally and go to the main screen
* long press `ENT` key and select option `Statistics`
* scroll to the screen `DEBUG`
* long press `MENU` key, the warning will apear.
  * pressing `EXIT` key will abort the test.
  * pressing `ENT` key will **preform the test**. 

Result of the test:

The radio should reset in about one second and then turn or again and resume the operation without any switch/pots/throttle warnings. The sound will also not work. There should be an asterisk icon beside the clock indicating that the radio was reset by the watchdog. After the normal power off, power on sequence the radio should work normally (warnings and sounds back, asterisk gone).

If the radio is powered off by the test, the the test **FAILED**.

Please report back the result of the test and include:
* result
* any mods done to the 9XE radio (like the C100 cap for RTC, etc..)

Here is the bin file for the testing: http://downloads-21.open-tx.org/test/opentx-9xe-2.1.2-watchdog-test.bin

lichtl, 2015-08-23:
Hi,

just performed the testing with your provided bin file. (I'm uploading a video to youtube and share the link later)

X9E000017A01
E02 BE MS
17/07/2015
Stock Radio (no RTC Mod or any changes)

Test: **FAILED**

Right after pressing enter the radio turned off. I've waited couple of seconds but it require to keep the power button hold (as normal). Once it then start I'll saw the icon on the top right and audio was disabled. It then just required once more power switching in order to return into normal state.

projectkk2glider, 2015-08-24:
@lichtl thanks for the test. Unfortunately your result proves that something is not right. The watchdog reset is not working as planned. This is bad.

@bsongis @kilrah @MikeBland what to do? Is this a hw bug (missing cap on power fet gate) or a sw one? 

The test also fails on my 9XE board, but my board presumably does not have the cap installed yet. The production radios should have it.

MikeBland, 2015-08-24:
I've just loaded the test code on my prototype board (not got a real X9E (yet?)). My board has a cap on the FET of the reset circuit. On triggering the watchdog reset, the board just switches off.

I've 'scoped the reset signal. It stats at 3.3V, drives low (very short spike), then rises (RC charge) to 3.1V in about 7mS, then the it starts dropping. At about 2.6V (11mS after first going low), it drives low again, holds low for about 2mS, then starts rising again (RC charge), but only gets to about 1.3V then it decays away as though the supply voltage is dropping.

I monitored the MCU_PWR_ON signal and once it goes low, it stays low.

projectkk2glider, 2015-08-24:
I will take a closer look at the startup code. Perhaps we look for a power button pressed signal, which is not present when wd reset happens.

MikeBland, 2015-08-24:
I've put my ersky9x firmware back on and re-tested watchdog resets (I have a debug command on the serial port for this). All works fine and it restarts straight back into normal operation.

The reset signal spikes low as with openTx, but simply rises back up to the full 3.3V after less than 10mS.

projectkk2glider, 2015-08-24:
Mike thanks a lot for your test. This proves that the bug is indeed in OpenTX. Which is a good thing, it means we can fix it fast. I will have a look now.

projectkk2glider, 2015-08-24:
Found it!

projectkk2glider, 2015-08-24:
Ok a bug was found and fixed. Now we need another round of tests. The new firmware (with included fix) is here http://downloads-21.open-tx.org/test/opentx-9xe-2.1.2-watchdog-test2.bin Same instructions as before.

@lichtl If you have the time, could you test the new version? Thanks.

projectkk2glider, 2015-08-24:
With this build even my second version of 9XE board works, both the watchdog reset and an exit from the OpenTX Bootloader. Hurray :laughing:

projectkk2glider, 2015-08-24:
@bsongis should we recommend to 9XE users not to fly with 2.1.1? This looks like a serious bug to me. And I think we should push out 2.1.2 ASAP if my fix proves to be good.

lichtl, 2015-08-24:
@projectkk2glider sure just performed the test.

Test: **PASSED** 

looking good now. Straight reboot with disabled audio.

MikeBland, 2015-08-24:
I'm not sure why you needed to change bootstart().
These three lines:
RCC->AHB1ENR |= RCC_AHB1ENR_GPIODEN; // Enable portD clock
GPIOD->BSRRL = 1;
GPIOD->MODER = (GPIOD->MODER & 0xFFFFFFFC) | 1;
are those I have in my bootloader and they work fine.

I'm also unsure why you need to make them conditional on the type of reset, surely you just need the power turned on at this point, whatever the reason you have started.

I would also question using "RCC->CSR & RCC_CSR...." in the #defines. After reading the reason for reset, I thought you needed to clear the bits in the RCC->CSR register, or they stay set for the next reset. This means you should red the register and store it, then clear the register, and use the stored value to determine the type of reset.

projectkk2glider, 2015-08-24:
>I'm not sure why you needed to change bootstart().

Well it obviously did not work. 

>I'm also unsure why you need to make them conditional on the type of reset, surely you just need the power turned on at this point, whatever the reason you have started.

No need to turn on the power if it is a normal start. The user is pressing the power button and providing a FET drive.

The whole port initialization and MCU_PWR_ON pin setting is done anyway again in the boardInit(), more precisely in pwrInit().  In case of watchdog reset that second initialization is obviously too late. 

>... I have in my bootloader and they work fine.

Why it is working in your version if a mystery. 

>I would also question using "RCC->CSR & RCC_CSR...." in the #defines. After reading the reason for reset, I thought you needed to clear the bits in the RCC->CSR register, or they stay set for the next reset. 

This is not how the board is behaving for me, but I will check this in the reference manual.

projectkk2glider, 2015-08-24:
@MikeBland I am out of ideas. Take a look at the last four commits https://github.com/opentx/opentx/compare/fee55eb91c...7c12287783 some are working some are not. Any idea what is happening? Perhaps some timing issue or what?

projectkk2glider, 2015-08-24:
Found it! From the stm32f4 errata sheet:
```
Delay after an RCC peripheral clock enabling

Description

A delay between an RCC peripheral clock enable and the effective peripheral enabling
should be taken into account in order to manage the peripheral read/write to registers.
This delay depends on the peripheral’s mapping:
• If the peripheral is mapped on AHB: the delay should be equal to 2 AHB cycles.
• If the peripheral is mapped on APB: the delay should be equal to 1 + (AHB/APB prescaler) cycles

Workarounds
1. Use the DSB instruction to stall the Cortex-M4 CPU pipeline until the instruction is
completed.
2. Insert “n” NOPs between the RCC enable bit write and the peripheral register writes
(n = 2 for AHB peripherals, n = 1 + AHB/APB prescaler in case of APB peripherals).
```

projectkk2glider, 2015-08-24:
So in the end the bug was in this piece of code:
```c
RCC->AHB1ENR |= RCC_AHB1ENR_GPIODEN; // Enable portD clock
GPIOD->BSRRL = 1;
GPIOD->MODER = (GPIOD->MODER & 0xFFFFFFFC) | 1;
```
There must be 2 NOP instructions between the first and second lines.

MikeBland, 2015-08-24:
This code works for me as it is, and, I believe worked originally during development.
I think you changed to a different compiler revision after this was working, and the code generated is different. My code compiles to writing the AHB1ENR register, then reads the address for the BSRRL before writing to it. The address is read from a flash location and this probably causes the required delay.
I've 'scoped the signals using my code, and your test2 code. My code simply drives the soft power signal back high, about 2.2mS after the watchdog reset.
Your code actually drives the soft power signal low, then the pullup you enable pulls it high, which is why it does restart after a watchdog reset.

I think there is another bug in the final code. On the X9E, one of the trims of interest is on port G, but the code doesn't enable a pull up on port G.

projectkk2glider, 2015-08-25:
In my last commit the port should be driven correctly, it is the same sequence than before, just delay added.
```c
void _bootStart()
{
#if defined(REV9E)
  RCC->AHB1ENR |= RCC_AHB1ENR_GPIOCEN | RCC_AHB1ENR_GPIOGEN | RCC_AHB1ENR_GPIODEN;
#else
  RCC->AHB1ENR |= RCC_AHB1ENR_GPIOCEN | RCC_AHB1ENR_GPIOEEN | RCC_AHB1ENR_GPIODEN;
#endif

  // these two NOPs are needed (see STM32F errata sheet) before the peripheral 
  // register can be written after the peripheral clock was enabled
  __ASM volatile ("nop");
  __ASM volatile ("nop");

  // Turn soft power ON now
  GPIOD->BSRRL = 1;                                  // set PWR_GPIO_PIN_ON pin to 1
  GPIOD->MODER = (GPIOD->MODER & 0xFFFFFFFC) | 1;    // General purpose output mode

  GPIOC->PUPDR = 0x00000004;
  GPIOE->PUPDR = 0x00000040;   
     // << MISSING PORTG PULL-UP CONFIGURATION
```

You are right about the port G pull up missing.

projectkk2glider, 2015-08-25:
In fact with the current code the OpenTX Bootloader can sometimes be started with only the right trim pressed. That is the consequence of missing port G pull-up.

MikeBland, 2015-08-25:
To help with the RTC reset problem, in the power off code, I use this:

    // It doesn't work correctly, if we press long on the pwr button, the radio restarts when the button is released
    PWR->CR |= PWR_CR_CWUF;
    /* Select STANDBY mode */
    PWR->CR |= PWR_CR_PDDS;
    /* Set SLEEPDEEP bit of Cortex System Control Register */
    SCB->SCR |= SCB_SCR_SLEEPDEEP_Msk;
    /* Request Wait For Event */
    __WFE();

To stop the problem with restarting if the power button is held on too long, just add this:

#if defined(REV9E)
	while ( pwrPressed() )
	{
    wdt_reset() ;
	}
#endif

  pwrOff(); // Only turn power off if necessary

before the pwrOff() call.

Mike.

projectkk2glider, 2015-08-25:
I've just built commit f0f951ca6bb8d0e545da1c42b2dff6a40339cd5c (it is the stock code with only the watchdog test option added, no modification of startup code  yet) on the official compilation server to see if compiler there produces a good watchdog reset code. **It does not** so all currently released versions of 2.1.x for 9XE are affected.

I will now focus on cleaning up my solution and testing it in various compiler versions. After that I will create a clean pull request. I will also look into the Mike's proposed code for RTC issue at the same time.

projectkk2glider, 2015-08-25:
With the latest commit 3e383bd6642b220ef8c07a4922bf091b7b7d5583 all three complier builds work:
* official compile server: 4.7.4 20140401 (release) [ARM/embedded-4_7-branch revision 209195]
* my 4.7: 4.7.4 20130913 (release) [ARM/embedded-4_7-branch revision 202601]
* my 4.8: 4.8.4 20140725 (release) [ARM/embedded-4_8-branch revision 213147]

commit 536e07b7 (2015-08-24): Fixes #2681: the PWR pin was not turned on correctly in _bootStart(). Watchdog reset now works on 9XE, the Bootloader exit also works.
commit e83b7c3a (2015-08-26): Fixes #2681: the PWR pin was not turned on correctly in _bootStart(). Watchdog reset now works on 9XE, the Bootloader exit also works.
Trim switch pull up not enabled for 9XE in _bootStart().
Added watchdog test option in DEBUG screen (long press MENU key), use WATCHDOG_TEST=YES compile option to enable it.
closed 2015-08-26 by projectkk2glider