opentx/opentx #2684: Companion 2.1.1: Lua custom script returning 100 is displayed at 0 in telemetry screen

fred99, 2015-08-09:
I run this trivial programme under Companion 2.1.1
-- LUA custom script
-- SD\SCRIPTS\MIXES
   
 local outputs = { "out" }
    local function run() 
      return 100 *10.24
    end
    return { run=run, output=outputs }

[image]

[image]
Should be 100:
[image]
This trivial progamme works fine on my Taranis updated in 2.1.1.

projectkk2glider, 2015-08-15:
Confirmed, fix coming up..

projectkk2glider, 2015-08-15:
It caused a segmentation fault in Companion actually!

closed 2015-08-15 by projectkk2glider

fred99, 2015-08-17:
Wow!
Companion seemed imperturbable.

commit 4d83dd2c (2015-08-15): Fixes #2684: Lua model scripts outputs not working in a simulator