opentx/opentx #2688: Vector and telementry not working together v2.1.1

MRTJ30, 2015-08-10:
Hi I have raised the issue before this new version but I was told you were aware of it and will be resolved in this new firmware update. Maybe other priorities were addressed in the upgrade rather than this issue everyone will have when trying to use Vector Eagle Tree in conjunction with the X8R receiver to display the Vectors telemetry on the X9D. Please confirm if this is still in fact a issue with the popup dialogue box that will not go away ( telemetry boxes are full) and press exit to exit but nothing happens.

MRTJ30, 2015-08-10:
Positive side I love my Radio and the great work you guys are doing with it

projectkk2glider, 2015-08-10:
Have you checked the new option "Ignore IDs". This should fix your problem. 

Duplicate of #2538

closed 2015-08-10 by projectkk2glider

MRTJ30, 2015-08-12:
You are a legend thanks mate. Reflashed it again and checked ignore instance and it works beautifully. Sent from Yahoo Mail for iPhoneOn 10/08/2015 7:36:11 pm, Damjan Adamic wrote:Closed #2688.