opentx/opentx #2690: EEPROM OVERFLOW on 9xr

mrpreciado, 2015-08-10:
i keep getting this warning EEPROM OVERFLOW ,i just have a couple of models saved and still displays the error
[image]

kilrah, 2015-08-10:
This does not look like an OpenTX screen but like the original firmware (er9x). It should tell you how much free space there is on the model selection screen if I remember well.

mrpreciado, 2015-08-10:
[image]
it looks like if it had no memory ,with just 2 models it displays it doesnt have enough  memory and when there are no saved models i get like 160 free,do you have any recomendation to solve this or what sould i try tnhks

bsongis, 2015-08-10:
This is not OpenTX.
OpenTX would show you 7 models in this last screenshot.

bsongis, 2015-08-10:
Also it seems to me that the screen index has a display problem (a white column between / and 11)

closed 2015-08-10 by bsongis

kilrah, 2015-08-11:
You should ask your question there: 
http://openrcforums.com/forum/viewforum.php?f=70