opentx/opentx #2693: Auto-offset for Frsky variometer on per default

fraternl, 2015-08-11:
In the telemetry options the Frsky-altimeter needs to have "auto-offset" enabled for proper behaviour. It would be nice if this is set at discovery.

Maybe this is the proper settings for all barometers

kilrah, 2015-08-11:
It's not a "need", it's personal preference. Was discussed and rejected not even a week ago.

fraternl, 2015-08-11:
It might be that I'm mistaken, but these variometer can't possibly give any sane value out of the box without this option selected, can they?

kilrah, 2015-08-11:
Or you may want a manual offset to see variation.

griebd, 2015-08-11:
@fraternl, I do not disagree with you, most people will set auto-offset and use this way... BUT that doesn't mean you can't use without it...
what @kilrah sad in hist comment is that you need to calibrate it everyday, maybe even every hour.... but you can calibrate it to your local altitude and have a valid ASL (above sea level) value.

fraternl, 2015-08-11:
I don't believe such a calibration is possible unless he means giving it a manual offset each day... (every 15 minutes)
An impractical way to achieve something that most people are not even interested in.....

A reset to 0 IS something that can be done by using some buttons or like I do using a special function.

Why hinder people that use it for relative height when it's already impractical to use for absolute heights...?

bsongis, 2015-08-11:
I tend to agree that we should have auto-offset enabled by default on all altitude fields as today all sensors are relative (or wrong if absolute)

kilrah, 2015-08-11:
OK, fine for me. In the usual philosophy I just always favor having as little "pretreatment" done as possible unless really necessary, and what the sensor gives should be what I see by default, if I then want it processed then I enable that myself.
Also kind of "forces" users to understand what they're doing and why.

commit 900b53f1 (2015-08-11): Fixes #2693
closed 2015-08-11 by bsongis