opentx/opentx #2694: More explicit names for |X| and |f| functions

pindeslandes, 2015-08-11:
Taranis.
It’s often difficult (at least to me) to remember what |X| and |f| functions actually do without referring to the manual because I find their names quite misleading.
Let’s take a mixer: y = a.x+b where x = input, a = weight, b = offset.
|X| doesn’t give y =  a.|x|+b, as it might be understood, but y = |a.x+b|
|f| doesn’t give |y|, as it might be understood, but Sign(y) (-100 when <0 and +100 when >0).

Maybe better names could be found, like |y| instead of |X| and Sign or -o+ instead of |f|.

projectkk2glider, 2015-08-15:
You are wrong about |X|. It is working like `y = a.|x|+b`. Try it with a negative weight or offset to prove. So that to me looks OK.

Now the other problem. The |f|  works like `y = sign(x)(-100 when <0 and +100 when >0 regardless of weight) + offset`. 

The whole range of `f` functions is a little strange. Not sure what their purpose originally was. Maybe it's functionality  was inadvertently lost or changed somewhere along the way. Can somebody chime in on what the functions are supposed to do:
 * f > 0
 * f < 0
 * |f|

tomkfly, 2015-08-15:
f > 0 and f < 0 are very useful for setting differential on the likes of Ailerons
[f]  makes  any value above or below offset travel equally to the full  set weight.

pindeslandes, 2015-08-15:
@projectkk2glider I am afraid you are right about |X|, I can't reproduce what I thought I had seen! Sorry.

closed 2020-07-07 by stale[bot]