opentx/opentx #2699: Running short of logical switches!

pindeslandes, 2015-08-12:
Taranis.
To me, the number of 32 logical switches appears to be quite a limitation to the power of OpenTx.

*I find it particularly true for training where, in addition to the warnings and safety functions of the plane itself:*
*- I want to be sure the student is ready to take control before giving it to him.*
*- I have two students, each of them having his own profile (rates…), and I want to be able, in flight, to tune their profiles and to shift from one student to the other, with safety functions to avoid messing up.*
*- I can choose to delegate or take back the throttle in flight, but I want to align the two throttles values (student and teacher) before, and to announce the student or teacher how much he has to move the stick.*
*All that functions need quite a number of LSs. Too many LSs, I’m afraid: at the end, I need to work around with mixers to realize some logical combinations!*

It would certainly be useful to extend the number of switches.

If not possible, maybe a compromise could be to have the additional AND switch of each LS not to be fixed to AND but to be chosen among AND, OR, XOR and even Sticky (in that case, Sticky would have its two inputs identical).

projectkk2glider, 2015-08-15:
Maybe you should use a Lua script for your purpose and with it free some of the Logical switches. 

The number of switches is a compromise. We can not expect everybody to be happy bu ti works for most of the users.

pindeslandes, 2015-08-15:
A Lua script would certainly be a solution but I am not familiar with scripts (like most regular users I believe) and logical switches can be modified at the field without a computer.
Maybe one day, I will try to learn about Lua...

fraternl, 2015-08-16:
I took the effort to make the use of switches more efficient and ended up with 10 logical switches. Today however I came up with a new idea for launching my DLG's and I ran out again....

Another idea for ending up with less LS's is having 2 conditions for a special function. Now you have to create an extra LS for this. If this condition is only needed for 1 SF it increases readability IMHO.
The relation between these 2 booleans now become apparent in that special function.
Otherwise this relation is not apparent at all.

mrabault, 2015-08-17:
fraternl, you have to understand that you are not special, you do not "deserve" the devs' attention  and they shouldn't be here to cater to all your needs. Instead, they try to cater to most user's needs.

That being said, if some users want more logical switches it could be done via a checkbox in Companion's preference, like mixermon but instead extendedls

bsongis, 2015-08-17:
More logical switches with our current EEPROM is difficult. Unless we stop
storing models on EEPROM and use only SD card...

pindeslandes, 2015-08-17:
I don't know if my last proposal would need less room than more logical switches would: allowing the additional condition of each LS not to be fixed to AND but to be chosen among AND, OR, XOR and even Sticky (in that case, Sticky would have its two inputs identical).
I mean that one:
[image]
By the way, that SD card idea seems quite good to me, from a user's point of vue, because I don't know any user with no SD card in his Taranis.

One more thing: if I had the choice, I would prefer 64 LSs + 32 SFs rather than 32 LSs + 64 SFs because, with the possibility to automaticaly read files like SA-up.wav or Mode_name-off.wav, we actually have more than 150 special functions available.

fraternl, 2015-08-17:
Bertrand, what about the idea of having 2 booleans for each SF?
It would also improve the readability of a setup

fraternl, 2015-08-17:
@mrabault  You apparently didn't get my remark....

I'm involved in testing and trying to improve OpenTX and this has nothing to do with how you see it. (Devs have to cater us greedy users) If you really think that way then this says more about you than about me.

kilrah, 2015-08-17:
> I agree, but your heart should lie with those that want to get the most out of it.... Not the mainstream users

So we have to implement something detrimental to 99% of users because you and a couple of others want it? Nice thinking.

fraternl, 2015-08-17:
Pfff  don't take it that serious...
It's about making real use of it.. which should be encouraged.
It's a feature that's there and we're using it.
It's not a 1%   many more hit that ceiling...  
There are enough curves, enough flight modes, enough special functions, enough models, enough mixes, but many can't cope with the switches that are available...

If it's not possible, then it's not possible.
I'm only suggesting things that could make it possible. If they wouldn't work, then so be it...

kilrah, 2015-08-17:
> There are enough curves, enough flight modes, enough special functions, enough models, enough mixes

You're again seeing things from your own perspective - because even if they're enough *for you* we have previously had complaints that there were not enough of pretty much each of those at one point or another from other people with special applications ;)

fraternl, 2015-08-17:
"You're again seeing things from your own perspective"
Why wouldn't I?

It's up to you to decide if things are worth doing taking into account what's possible...
I think you and your colleagues are quite capable of doing that.

You just think I want to be catered for every need and have a problem with it if some wish doesn't get granted.

I wish this discussion was about trying to solve it and/or telling me that the proposal I made would not make any memory free.

pindeslandes, 2015-08-18:
Out of the subject but is there a way (and would it be a good idea) to do some kind of user's opinion pool about undecided issues like this one, when things have been discussed long enough to help every one to make his mind?

LapinFou, 2015-08-18:
I like very much the idea of pindeslandes: be able to choose any boolean operator instead the fixed "AND switch". Unless there is still an EEPROM storage issue.

bsongis, 2015-08-18:
I think it's possible from the EEPROM point of view. Selecting OR / AND
costs only one bit per line. How would we fit this on the LCD is now the
problem, the line is short...

LapinFou, 2015-08-18:
Got it. Maybe, we can used the boolean symbol ?
"&" or "." for AND switch
"+" for OR switch
But, it may be confusing for average users... :(

fraternl, 2015-08-19:
And what about my suggestion of having 2 anded switches in the special function section? Often (in my setups) I need to make a special line for that.
Not only is that line not needed then, but it also could be on the next page where there was still some room left
If it's done in "special functions" the relation between those 2 switches become clear

fraternl, 2015-08-19:
For the other solution (switchable AND/OR) the header could change...  Maybe flashing if it is "OR" and the switch itself in italics or underlined

bsongis, 2015-08-19:
@fratern Would you do a screenshot to show us if your idea is something we
can do with our LCD width?

fraternl, 2015-08-19:
I could do that this evening

A long press could toggle between AND / OR
The OR (being an exception) could be emphasized by having it in reverse.

In the first row it can be seen what it currently is.

In case of "OR" it could also alternate between "OR" and its value if not selected.

lshems, 2016-01-19:
setting up 8 flight modes on 2 physical switch combinations will already cost you 8 LS, if you don't want to use the order of the assigments as a setup functionality.

I avoid that, because it is a 'hidden' feature.

Then, since special functions can only trigger on a LS or physical switch, and not on an input (why not?), we need another 2x6= 12 LS to announce the functions assigned to the non default physical switch position if they are conditional to some other setting.

I really miss some extra LS

Alternative for me would be to have the special functions trigger itself on a condition, just if they where Logical switches themselves.

So SF1 is triggered when Input1 > 2, or when SAup and SBdown, in a sort of "direct" mode

bsongis, 2016-01-19:
Done in 2.2 (next branch)

closed 2016-01-19 by bsongis

lshems, 2016-01-19:
Thanks, it will save my day. I can wait, but really needed to know if it would be coming.