opentx/opentx #2700: Multiple voice choice

pindeslandes, 2015-08-12:
Taranis.
It would be nice if it were possible to choose the voice you want to use for ‘Play value’ special function.
The typical use I see would be for training, in order to have the usual sweet female voice for the teacher and a strong, authoritarian male voice for the student (who stands a few meters away from the teacher’s Tx in a potentially noisy environment and needs to obey!).

projectkk2glider, 2015-08-15:
Is this on one radio and one model? Could you have two models on the same radio, basically a duplicate but one using other voices?

pindeslandes, 2015-08-15:
Yes, one radio and one model during the same flight. The idea is that the student would be sure to ear the value and would have no doubt that the annoncement is for him.
For the moment, all I can do is this kind of thing:
- Strong male voice: "Reduce the throttle by..." (custom file in the SOUNDS/language directory)
- Sweet female voice: "...four..." (SOUNDS/language/SYSTEM directory)
- Strong male voice: "...steps."

Sounds funny.
OK, there the student is aware that the radio is speaking to him, but He would answers: "How many?"

griebd, 2015-08-15:
who is the instructor? in a few years we would need no instructor, the radio will do it by himself... it will even fly the model for us... wait a second, there are lost of autonomous models already, I'm getting too old....
do NOT get angry, just making fun...

BUT, seriously, instructions for a student should be done by the instructor, not the radio... at least that is how I do it! but different people do things different, so...

pindeslandes, 2015-08-16:
OK, I get your point but let me describe the whole sequence:
1- The student is controling the model but not the throttle. I want to give him the throttle, as He feels OK for that, but I wish to align teacher's and student's throttles before, to make it smooth.
2- I ask the radio to tell the student the way He must move his stick.
3- Then the radio says to the student: "Reduce the throttle by four steps" (for example). That helps the teacher a lot because the teacher can't know the stick position of the student. Only the radio knows.
4- The student gently executes this instruction.
5- As soon as the throttles are aligned, the radio says to the student:" Stop",
6- If the student actually stops moving, the radio tells him: "You have the throttle" and it automatically gives the throttle to the student. Of course, this information is very impotant both for teacher and student.

So, I am the teacher but my amazing OpenTx flashed Taranis makes it easier and safer for me and my students.

projectkk2glider, 2015-08-16:
Ok you have a point there. Currently I have no idea how to elegantly solve such case. Lets wait for others to chime in.

LapinFou, 2015-08-18:
The idea of pindeslandes to align the sticks are nice, but I still don't see the need to have multiple voice choice. If you are creating some special function for the training mode, you can also created some custom voices for those special functions. In others words, what you described can already be done with the current version.
Or did I missed something?

pindeslandes, 2015-08-19:
Yes, @LapinFou, I believe that what you missed is that, still in this example, the number of steps the student must move his stick is given by the ‘Play value’ special function, which uses files from the SOUNDS/language/SYSTEM directory, where you can't obviously have two files (sweet female voice and strong male voice) with the same name. So you can't choose between two voices in flight.

kilrah, 2015-08-19:
What we don't see is the need for separating those voices in the first place. You could simply make a "Listen now!" Or "Student that's for you!" prompt that you play before something for him is said when required.

pindeslandes, 2015-08-19:
@kilrah you are not wrong, it's somehow what I am doing already while starting student's annoncements with the strong male voice.
My point is just that having two system voices would be more elegant. And so much cooler!

LapinFou, 2015-08-19:
@pindeslandes 
OK go it! Thanks for the clarification.

pindeslandes, 2015-08-22:
Maybe a smart way to solve this issue (from a user’s point of view, of course, I have no idea of how much it would be difficult to implement) would be to have the additional voice in a SYSTEM2 directory that would be called by a ‘Play Value 2’ special function?

bsongis, 2015-08-22:
Perhaps a new function SetVoice which would propose directories XXX and YYY
if en-XXX and en-YYY are present?

pindeslandes, 2015-08-22:
In this case, from what I understand, this function would be a model setup function and you woudn't be able to choose the voice you want in flight, would you?

kilrah, 2015-08-22:
You'd have to add a SetVoice SF before any voice you'd want played with a voice different than the previous one. Might use a lot of SFs in your case, but would work.

pindeslandes, 2015-08-22:
Yes. Thanks for the clarification.
Not the simplest way for users but the easiest solution to develop (I guess) and that would certainly fit most user's need (would fit mine anyway!).

fraternl, 2015-08-24:
That XXX could well be just 1 letter in the alphabet.
I already purposes this here:

A SF called "voice suffix" could be empty (default) or a-z

If it's a then the base sound folder becomes /sounds/en/a/
It would stay that way until another SF changes it to another letter or empty.

A radio-wide default suffix could also be set and then a wish I had before could be fulfilled too.

projectkk2glider, 2015-08-24:
This is not very trivial to implement. Remember we reference (search for) all the available sound files at the model load. With this enhancement the number of files could bi much bigger.

closed 2019-11-15 by stale[bot]