opentx/opentx #2702: Timer (2) stops counting after around 9 hours

Gerporgl, 2015-08-13:
Not sure if this is a one shot issue, but I have a persistent timer 2 to count the time how long my TX battery last, and it stopped after 9 hours something (sorry should have written down the exact value before before resetting).
Is this a known issue?
I am running the latest stable version of opentx 2.1.1

It looks like it should be able to go to 24 hours or even 99 hours, but it stopped for some reason. I powered it down and on and it was still stuck at 9 hours. It is only after resetting it that it restarted counting from 0.

It would be great if this could count to more than 9 hours, since with newer lipo battery in FrSky Taranis it last very long and I find it useful to see the battery run time.

bsongis, 2015-08-13:
Yes it will stop at 32767 / 3600 = around 9 hours
This should be changed at least on ARM boards

kilrah, 2015-08-13:
For your purpose you might be better off using the system timer you can see on the statistics page. I doubt you only have one model so using a model timer for that is a bit pointless...

closed 2015-08-13 by bsongis

Gerporgl, 2015-08-14:
I see. In fact I got the Taranis only 2 weeks ago, and I haven't yet transfered all my models from my previous radio. I did not notice the system timer in the statistics menu before. Good to know, thanks for the info.

This is amazing what we can do with OpenTX compared to what I could do in my Aurora 9! I can't believe the difference.

commit 9303361d (2015-08-13): Fixes #2702