opentx/opentx #2705: There could be a way to specify a previous fw version

rafabene, 2015-08-14:
I moved recently from 2.0.x to 2.1.1 in my 9xrpro and that broke the sound. I wish to be able to use 2.1.0 in companion.

kilrah, 2015-08-14:
2.1.0 was broken too.

projectkk2glider, 2015-08-15:
@rafabene explain more what do you want exactly.

rafabene, 2015-08-17:
I would like that companion be a able to select previous versions (downgrade) a firmware. This is a feature request!

projectkk2glider, 2015-08-21:
Not sure the whole firmware compilation server thing would be capable of such an arrangement. I guess it would be too many firmware images to keep. The @bsongis will decide.

Still the long standing recommendation is valid: keep a local backup of the old downloaded firmware versions to use in such cases.

fraternl, 2015-08-22:
Couldn't the system keep all the once requested flavours in some folder nicely arranged?
It's not exactly what he wants, but could be enough.

kilrah, 2015-08-22:
No, if we do something about that it should be done in a similar way to nightlies. 
But I'm not for it, or would restrict it to only the previous version, nothing further back. But then we need a way to disable that should the previous version have a serious bug etc, more work and maintenance.

closed 2019-11-15 by stale[bot]