opentx/opentx #2706: Heavily distorted audio

frank16, 2015-08-14:
I recently bought a Taranis 9d+ and noticed the audio was really bad, so I turned on my oscilloscope and had a look. I found out the 'Welcome to OpenTx' message to be really over modulated (heavily clipped). I couldn't find the audio circuits (on the backside of the V0.3 mainboard maybe??) but
I traced it down to the 'tada.wav' file (using audacity) which was already clipped!

I haven't tested all voices in audacity, but at least the english-amber and english-scottish-taranis voice have it, the 'english-kilrah' voice does not (seems to be softer in general for what I checked). 

I then installed the kilrah voice and was pleasantly surprised. The quality isn't great, which is to be expected with that tiny speaker embedded in plastic, but I see no obvious distortion on the oscilloscope even with the volume to max for both 'volume' and 'wav' parameters at the same time. The sound does get more distortion but that feels to be more related to the speaker and its housing.

rafabene, 2015-08-14:
I think this is duplicate of #2665 which is already fixed. I'm wondering how can I get access to the nightly builds to test it

kilrah, 2015-08-14:
The stock voices are autogenerated and sound extremely "robotic" in the first place, just listen to them on the PC and you'll see they're no better. They're also output at full volume, so you might need to reduce wav volume a bit so they don't clip.

The english-kilrah pack was made with a completely different tool with much better sound quality, but needed hours of manual preparation. It's also gone through a few iterations of finding a proper equalizer setting to come out pretty well on a Taranis Plus speaker. Volume in the file is about half as a result.

Anyway not an issue, just a question of how the sound files are made.

closed 2015-08-14 by kilrah

frank16, 2015-08-14:
Well then at least taranis owners can upgrade their stock voice by going to 
eng-kilrah (it's not a trivial difference in quality to my ears).
Something gained...
Strange though it isn't the default if it so much better. Am I missing 
something?

I just realised that although the sound files looked clipped to me on an 
oscilloscope they could actually be hard dynamic range limited at 100%. I 
should have zoomed in more closely. Still sounds bad though...

But then how does one fix say the 'Amber' voice so it is softer but 
dynamically correct and better sounding. You can't adjust the .wav file,  
it has to be regenerated.
Is that in another project?

kilrah, 2015-08-15:
> Strange though it isn't the default if it so much better. Am I missing something?

Simply because we're sure the autogenerated ones match the code if changes are made, and mine wasn't available on day one.

> You can't adjust the .wav file

Of course you can load that in a sound editor and apply whatever changes you want. The Amber pack is a real recorded voice, the wife of someone on RCGroups. A few new sounds are however again synthetic as the creators didn't show up and/or update it recently.

frank16, 2015-08-16:
Sure, I understand you can edit a .wav file, but if it's clipped or hard 
dynamics limited, you can't get back what isn't there anymore...

But I finally get the point. You need to autogenerate to be able to 
guarantee support for new texts.

May I ask what software you guys use for that and where the input texts 
are? I may want to play with that a bit myself.

For the moment I'm working on the hardware side. What a lovely combination, 
taranis and opentx. I'm in hackers heaven :-)
Maybe I'll finally build my own transmitter, have been wanting to do that 
since my highschool days in the '70s.... There have been several tempting 
projects over the years, but opentx with frsky arm based hardware is the 
one for me. Finally something fast enough. Can't wait to design a receiver 
extension with gyro and gravity sensors that will allow me to lock in a 
perfect turning rate on a scale model or bring a glider back to a 
predefined position when you lose it in a thermal... Anybody working on 
that that you know of?

Thanks!
Frank

kilrah, 2015-08-17:
> May I ask what software you guys use for that and where the input texts are? I may want to play with that a bit myself.

I believe they're made with the SAPI engine on Windows. See radio\util\tts.py