opentx/opentx #2707: Center beep on setup tab missing for third potentiometer

powercycle, 2015-08-15:
There is a P1 and a P2, but should also have a P3.  P3 appears in source listings as expected.

projectkk2glider, 2015-08-15:
More data please, which version, on the radio or in Companion?

projectkk2glider, 2015-08-15:
9XR does not have P2. The 9XR-PRO does have one and I can see the center beep selector in Companion 2.1.1.

powercycle, 2015-08-16:
Companion, 2.015.

9XR has three potentiometers, in the mixes tab for a SRC you can see all three, P1, P2, and P3.

In Setup tab, only P1 and P2 are shown.

Center Beep on simulator (and Radio) DOES show the third Pot.

So only seems to be missing on the Setup tab.

projectkk2glider, 2015-08-16:
Ok, I did not know that 9XR has three pots.

powercycle, 2015-08-22:
From all the pictures I've seen, the 9X has three as well.

Mike B.

bsongis, 2015-10-13:
I can't see the bug in current 'master', did someone already fix it?

kilrah, 2015-10-13:
The bug was tagged wrong, it's an issue only in 2.0!

bsongis, 2015-10-13:
Ok that's different

bsongis, 2015-10-13:
No issue in 'master', just fixed in 2.0. I close this issue.

commit 7284db85 (2015-10-13): Fixes #2707
closed 2015-10-13 by bsongis