opentx/opentx #2708: Switch that becomes true when going from 1 switch to another

fraternl, 2015-08-15:
I have asked in the forum how to create a switch that becomes true when I go from 1 switch to another.
I suspect it can be done with edge, but I'm doing too many things at the same time and have some difficult to understand the possibilities with it. I also suspect it needs to be more than a one-liner.

If so, I would like to propose a switch that becomes true when going, for instance to FM0, but only if the previous cycle was in FM2. 

If this can done with another one-liner I would like to know how and you can close this issue.

tomkfly, 2015-08-15:
I have just posted the following reply to your post on RC Groups.
"I've used L1 and L2 in the suggestion below, but you can of course use any free Logic Switches
Set up two logic switches
L1 / Sticky / FM2 / L2 
L2 / AND / L1 / FM0 / with a short duration, about (0.5 seconds)
Use L2 to trigger the sound in Special Functions.

I haven't been able to check it, but I'm pretty sure that should do what you want.

Tom"

fraternl, 2015-08-16:
I've implemented the 2-liner with Edge. It works, but it does take up 2 lines

The trigger, if it is ever implemented, could be called "Morph".
I do agree that such a trigger should have an "elementary character". I leave that up to you.
It has in a way

fraternl, 2015-08-25:
To clarify my feature request:

L1  Morph(FM2,FM1)

Would be thesame as:

L1    Edge(FM2, [0:0]) Duration(0.3s)
L2    FM0 AND L1

closed 2016-11-27 by fraternl