opentx/opentx #2710: Companion 2.1.1 Curves with micro seconds

rguptaca, 2015-08-15:
Having the Y axis optionally show microseconds will be useful. Or -100 to 100 where it is now and microseconds on the other side.

projectkk2glider, 2015-08-15:
There is no point. The curve output is affected by mixer weight, mixer offset and servo output range. The ms value would be meaningless. Reopen if you have a compelling argument against that.

closed 2015-08-15 by projectkk2glider

rguptaca, 2015-08-15:
So why have curves at all? Ill submit an image taken from. +360 video. A curve was used to get the correct miliseconds for 6 different modes. 
Keep in mind there are many of us doing things developers may not have thought of. I find this arbitary closing by just one person a bit disrespectful. You guys do an amazing job but would it hurt to see if anyone else has interest in this?

griebd, 2015-08-15:
maybe, just maybe, the right place to propose and discuss new ideas is a forum (there are links in OpenTX website).... if there are enough momentum there, the proposer come here and open the issue request WITH a link to that thread! It will have a lot more weight against the developers... ;)

For the devs: OpenTX user base is growing a lot, it's time to make a more organized workflow or it will turn out a mess, with lots of complains... I know, it's a free project, but why not make it a huge free project?! Linux is still there nowadays... ;)

anyway, just my 2 cents (again... ;)

kilrah, 2015-08-15:
Correct, discuss that with others first so they will be able to tell you this makes no sense. 
As Damjan told you the curve is only a multiplier that can be used at many different places and with various modifiers, which means there is no direct way to map a curve to a pulse.

rguptaca, 2015-08-15:
http://youtu.be/TRGZWFTCx4A
Here is one example where using ms would help. 
Granted there are more ways to achieve this.  But it shows how curves can be used. 

Anyway. As always Thanks for your time put into this project. Is there a new feature section in the forum or are new features mixed in with the rest of the posts. If there is I missed. My apologies. 

Rav

Have a wonderful day!

Ravi Gupta, PMP
Founder of Hike For Happiness
A Mental Health Fundraiser
http://hikeforhappiness.ca/
(519) 697-0796