opentx/opentx #2711: Companion 2.2.1 Model Wizard limited to 8 channels

rguptaca, 2015-08-15:
Select Plane
Standard flying wing
Throttle Ch 1
Aileron controlled by Ch 2 and 5
Flaps controlled by Ch 6 and 8
Airbrakes controlled by Ch 7 and 4
Tail elevator and rudder
Rudder Ch 3
Elevator Ch 3 (same as rudder)

projectkk2glider, 2015-08-15:
Duplicate of #2624

projectkk2glider, 2015-08-15:
Ok, the CH3 duplicate usage is a bug. 

But as explained in #2624 such complex setup is beyond the wizard.

griebd, 2015-08-15:
just wait a second... some clarifications:

@projectkk2glider, it's a companion issue, by the title, not radio firmware.... ;)

@rguptaca, companion 2.1.1, not 2.2.1 (there is no such)... ;)

griebd, 2015-08-15:
Was fixed in 2.0 already, probably missed to port to next at the time!

Thing is, the fix is only a warning so people know they should do things by them self... Again, the wizard is just a show up for newcomers. :wink:

rguptaca, 2015-08-15:
Where did I write 2.2.1? If I did apologies for causing confusion if any. 
For beginners maybe simplify the wizard to just TAER and maybe Gear with only AIL having a choice of 2 seperate or 1 channel. Most off the shelf planes have simple setups.  Or at the end mention the 8 channels have been exceeded. 
I did read the titles of every open issue before posting. 
Only trying to help.

griebd, 2015-08-15:
no problems @rguptaca, I thought it was a typo in the title and changed it... and the issue is valid as the fix was done only in 2.0.15 and not in 2.1 and later. (the issue for 2.0 was #2091)
This issue will auto close as soon as the fix get merged for 2.1.2.

PR #2712 (2015-08-16): Griebd/model wizard issue 2711
Fixes #2711.
Cherry-pick from 2.0. Was a fix for #2091 (#2143).
Probably missed to port to next at the time.
closed 2015-08-16 by projectkk2glider