opentx/opentx #2714: Strange behaviour in telemetry page

MicheleVilla, 2015-08-15:
Taranis (not plus)
opentx 2.1.2N418
Italian language
X8R receiver binded

Telemetry page
Screen1 
option: Value

When scroll parameter the showed value are
.....
.....
.....
Gv9
  OFF
ABSM
NO
(Blank)
(Blank)
RSSI
RSSI-
RSSI+
RxBt
RxBt-
RxBt+
SWR
SWR-
SWR+
-
*

If choise "NO"  in telemetry page is showed T1
If choise first Blank space in telemetry page is showed T2
If choise second Blank space in telemetry page is showed T3

Tx battery is no available

Possible is a translation bug?

Tanks

kilrah, 2015-08-15:
Yes there must be errors in the translation.

MicheleVilla, 2015-08-15:
I have found some difference from it.h.txt to en.h.txt
I will try to edit file later

projectkk2glider, 2015-08-16:
The Italian (and possibly others) have a lot of changes missing in `TR_VSRCRAW`  which is now assembled from multiple other strings.

projectkk2glider, 2015-08-16:
@MicheleVilla let me fix that first and they you can make your translations.

projectkk2glider, 2015-08-16:
@MicheleVilla It should be fixed now.

MicheleVilla, 2015-08-16:
Thank you for your fixes
Now I will start to fix other translations and test all

MicheleVilla, 2015-08-16:
Now seems all Ok

MicheleVilla, 2015-08-17:
Update:
Taranis  with X8R
firmware compiled from source  (today 17/08)
both simulator and firmware installed on radio

On Simulator is Ok but on radio the last option are
SWR
SWR-
SWR+
"-"
"+"

What means two last option?
are a bug?

projectkk2glider, 2015-08-17:
Yes, can you upload your eepe? Have you tried English language, does it have the same problem?

MicheleVilla, 2015-08-17:
Try with other memory an new model
all ok 
problem is only with a singol model
I check but I think there is no other bug
Do you want the eepe?

The telemetry parameter are memorized the first time radio is connected to receiver and remains in memory
Is possible to reset the data of telemetry ?
Or the procedure is different?

before post is better to make different test......

projectkk2glider, 2015-08-18:
>Is possible to reset the data of telemetry ?

If you mean to clear all detected telemetry values, then there is no such function avalilable (yet). @bsongis maybe we should add a menu option to delete all sensors?

bsongis, 2015-08-18:
Yes why not!

WZ9V, 2015-08-20:
Yes please.  It would come in handy when dealing with systems that goof up and fill up all the slots.  I just had it happen today with TauLabs and now I need to dump all the discovered sensors to test the fix.  Probably won't get used much once things are sortd out but for now it's almost a necessity until everything else gets updated.

bsongis, 2015-08-20:
Then your Taulabs firmware doesn't fulfill S.PORT protocol rules. In 2.1.1
you can ignore the telemetry ID, it will hide your sensor bug.

kilrah, 2015-08-20:
Just do the tests on a throwaway new model that you can just delete and make a new one until everything's sorted.

WZ9V, 2015-08-20:
The Taulabs folks have corrected the problem and it will be in the next version.  I tested their fix last night.  I believe the approach they used was to have the FC use one device ID.  Now I just get the expected values without flooding all the slots.  That branch is bleeding edge so I'll just disable telemetry until it's pushed out as a stable release.  Kilrah, that's the approach I took, just deleted the model made a new one and then copied it for the test so I could delete it if telemetry got flooded again.

projectkk2glider, 2015-08-21:
@MicheleVilla is this fixed by using the new model or not?

MicheleVilla, 2015-08-22:
@projectkk2glider 
sorry for delay
with new model is fixed
for me this issue may be closed
Thanks

closed 2015-08-22 by kilrah