opentx/opentx #2715: Throttle trim - Companion Simulator Error

driver33, 2015-08-15:
Comapanion v 2.1.2N418.
Throttle trim working as normal trim not throttle trim - only for slow rpm position.

pindeslandes, 2015-08-16:
I believe it has always been like this. There is a check box here to choose the way the throttle trim works: idle only or full range (sorry, it's in french):
[image]

projectkk2glider, 2015-08-16:
Can not reproduce. Provide more data, like which radio, maybe even upload eepe. Yo do realize that throttle trim idle only also affects the whole throttle range, just its effect linearly diminishes with the throttle position.

driver33, 2015-08-16:
Yeah, My mistake I think. Watching only INPUT with works from -125 to 125 (input (-100-+100) + trim(-25+25)) and didn't check channel output.

Closing this issue.

closed 2015-08-16 by driver33