opentx/opentx #2717: Trigger to know there is a valid link

fraternl, 2015-08-16:
Today I read about someone that used "A1 = 0" as a trigger to know that the receiver was off (2.0.x).
In that case he lets the Radio beep every 5 seconds.
If he puts his DLG in its prelaunch mode while that battery is still at 0 Volts it will sound an alarm...
It would save his model and prevent him from launching it.

Now in 2.1x the telemetry behaviour has changed. If no input is received from telemetry, the condition of switches involving them no longer become True.
This is a good thing IMHO, but now I can't come up with a trick to let the radio know the receiver is turned on (and there's a valid link).

projectkk2glider, 2015-08-16:
Why not use RSSI value?

fraternl, 2015-08-16:
Tried it.... same thing

projectkk2glider, 2015-08-16:
Sorry cant test this on a radio now, but are you saying that RSSI is not zero when the telemetry link is not active (RX off)?

fraternl, 2015-08-16:
No, it isn;t
Besides that, If it's a boolean (the same one as the one that triggers "telemetry lost / telemetry recovered") I can use it immediately in another LS without burning one of the available switches....

I believe no values are considered valid to trigger any response if no value is obtained from telemetry.

projectkk2glider, 2015-08-16:
Then we have a problem.

Mow06, 2015-08-18:
Use "RxBt > 1V ". or "A1 > 1V"  Goes true when telemetry available, Goes false when telemetry lost.

fraternl, 2015-08-18:
Thanks, good workaround...
Will use that.
You tested it I presume?

Still I hope we get a Boolean for telemetry active.
It would make a switch better readable, future proof and it doesn't take up a switch

bsongis, 2015-08-18:
Yes, we could easily add a set of "telemetry validity switches", and they
could be available in AND conditions.

fraternl, 2015-08-18:
Hi Bertrand, 
What do you mean by "a set of..." ??
I'm merely asking for 1 Boolean that gives me the status of the existing link.... True/false

projectkk2glider, 2015-08-18:
He means that in 2.1 each telemetry value has already a last update time value that can be used to determine if the value is current or a stale one. Using that as a criteria, ie if last update time is too old, then the value of validity is FALSE.

fraternl, 2015-08-18:
Without knowing that much of the source code I would go for the same thing that also triggers the "telemetry lost/telemetry recovered".

But that's out of my area and ask up to you..

fraternl, 2015-08-18:
Just noticed that an option to delete all the telemetry settings is in the make... It would be a shame if this would invalidate such a trigger in the logical switch section...

projectkk2glider, 2015-08-18:
What do you mean? That issue has no influence on this one.

fraternl, 2015-08-19:
As i wrote "not my area"...  :-)

BTW...
Is it correct that this morning's nightly doesn't include the fix? I don't have my own development environment, so I'm dependent on nightlies.

You're asking to test, but I can only test it if it's in nightly.

bsongis, 2015-08-19:
@fratern is it possible you don't mix issues discussions, it's more and
more difficult to follow.

projectkk2glider, 2015-08-19:
>Is it correct that this morning's nightly doesn't include the fix? I don't have my own development environment, so I'm dependent on nightlies.

Which fix? This issue so far has no fixes/changes to the code. If you mean some other issue please post there.

kilrah, 2015-08-19:
A quick look at the commit history helps seeing what's done about what.
https://github.com/opentx/opentx/commits/master

fraternl, 2015-08-19:
Sorry, i was confused.
At the moment I wrote that I was thinking of the issue with "edge". 
I was doing too many things at the same time

projectkk2glider, 2015-08-21:
I removed a lot of posts that were not relevant to this issue.

Now, on 2.1 we have a problem. There is no value that can be used in logical switches to determine if telemetry  is streaming or not. The same is true for Lua there is no way a script can find out if telemetry is streaming or not.

This functionality got lost, because of the new telemetry system in 2.1. 

Possible solutions:
* we expose `frskyData.rssi.value` which is set to zero when telemetry is not streaming
* (as Bertrand proposed above) we expose the result of  `TelemetryItem::isOld()` for each telemetry sensor/value.

Comments welcome. I like the second option, but it will increase the number of available fields in selection lists. We also need suggestion for a symbol for validity value. Currently we have:
* `name` for normal value
* `name+` for max
* `name-` for min
* for validity we need something new, my suggestion `name?`

griebd, 2015-08-22:
I like the idea to know if a specific sensor is valid or not (isOld()), BUT I think that having access to rssi.value is a lot more reliable. If we use a sensor we may have a misbehavior because that sensor is fault, for example, and we still have telemetry...

Mow06, 2015-08-22:
Don't' understand what this is all about. In 2.0.17 "A1=0" never goes true. I just tested it again.
However "A1>0" is true whenever telemetry is active and false when telemetry is lost.

2.1.1 does the same. What's changed?

fraternl, 2015-08-22:
I read this "a=x A1 0" in a post and tried it. It didn't work. So I assumed it was because of the 2.1.x
I'm not going back to 2.0.x anymore and had no reason to doubt that post.
That it didn't work in 2.0.x is new to me.

Yes, you are right. 
I can use the "A1 > 0" or "RxBat > 0" as I later found out after posting this issue.

I think the issue is still valid, albeit less urgent.
To create an alarm together with some action (put a glider in launch mode) I would need to create this line "RxBat > 0" first before I can use it.
A Boolean containing "link" could be used immediately without wasting a line in my switches.

The "trick" might also not work in some situations, although I can't think of any right now (on D-series)'

fraternl, 2015-08-22:
Name suggestion: "link"

projectkk2glider, 2015-08-22:
The issue of  "A1>0" working and "A1=0" not working is strange. if the first one is FALSE then the second one should be TRUE. Perhaps we have another bug in there.

Here is an explanation how things currently work:
* each time a valid RSSI packet is received the `frskyStreaming` counter is set to 1 second.
* if the counter `frskyStreaming` reaches 0 then the `frskyData.rssi` is set to 0 immediately. 
* other telemetry values are updated only if `frskyData.rssi` is bigger than 0. 
* the `frskyData.rssi` can also be set to 0 if the internal XJT module reports it that way. But the value itself is filtered so it wont reach the zero after the first packet.

So what happens when the RX is turned off:
* the `frskyStreaming` starts decreasing
* we might still get in a last A1 or A2 packet (before reciever dies completely) with reduced value.
* as soon as the filtered `frskyData.rssi value` reaches zero **or** when the `frskyStreaming` timeout elapses, other telemetry values stop getting any updates - they freeze.

So if A1 or A2 drop in such cases it is a merely a happy coincidence and not a sure way of telling that the telemetry has stopped.

So as @fraternl sugessted perhaps exposing the `frskyData.rssi` value as a `LINK` boolean value would be a good thing.

Mow06, 2015-08-22:
"So if A1 or A2 drop in such cases it is a merely a happy coincidence and not a sure way of telling that the telemetry has stopped."

Can you explain this?

I'm looking at a tx now 2.0.17.

A1 is 5.2V on the Tx.      "a1>0" is true.
Shutdown RX.
A1 is still 5.2V on the Tx, but "a1>0" is false.
power rx.
A1 is still 5.2V on Tx. "a1>0" is true.

kilrah, 2015-08-22:
Yes because when telemetry is lost all LSs based on telemetry values become false. Otherwise you'd be getting mad with low battery alarms etc sounding continuously when the receiver is off.

Mow06, 2015-08-22:
"Yes because when telemetry is lost all LSs based on telemetry values become false. Otherwise you'd be getting mad with low battery alarms etc sounding continuously when the receiver is off."

So....
"A1>0" going false is therefore a reliable way of detecting loss of telemetry.

And this:-
"So if A1 or A2 drop in such cases it is a merely a happy coincidence and not a sure way of telling that the telemetry has stopped." is incorrect as "a1>0"  will go false when telemetry is lost (not because of the value of A1, but because telemetry has stopped)?

projectkk2glider, 2015-08-22:
Kilrah is right, I completely forgot about telemetry based Logical switches becoming FALSE when the telemetry is not streaming. This has not changed in 2.1 so it should still work.

But as we saw above both of these switches become false when the telemetry is not received:
* A1>0
* A1=0

Why is that? Because every logical switch that compares a telemetry value with some other value becomes FALSE no matter what the actual result of comparison is in the case that the telemetry is not received.

As far as Lua is concerned the same principle is used (which also has not changed from 2.0). When the telemetry stream is not received the function `getValue()` returns 0 for every telemetry value.

fraternl, 2015-08-24:
The conclusion of all this is that we get a Boolean called "link"...   are we?

kilrah, 2015-08-24:
IMO to do it well we'd add a set of new items at the end of the switch list with one item for general telemetry then one for each known sensor.
Now this however causes more difficulties because switch names are limited to 3 chars long while telemetry item names have 4 chars.

Another option could be to only have an "LNK" switch, then use a new type of "Telm" logical switch for individual items...

In both cases this will likely have to wait for 2.2 due to format changes.

fraternl, 2015-08-24:
For me it would be enough to have that LNK boolean,
Having it for the telemetry sensors would be nice indeed, but not in that hurry for that...
So if the "telemetry sensor booleans" are stalling that, I would prefer to have those later.

fraternl, 2016-09-19:
I was setting up a failsafe on the Horus last night and noticed the LS "Tele".
Is this in fact the switch I requested here to test for a valid link?

I already used it in a global function and it gives a warning beep when "Tele" is false. 
I didn't have time to test if it worked as I expected.

kilrah, 2016-09-19:
Yes.

closed 2016-09-20 by kilrah