opentx/opentx #2718: Copy "outputs" from 1 model to another or....   partial pasting....

fraternl, 2015-08-16:
I fly 4 different DLG's with exactly the same program.
If I want to quickly implement these changes to other models I create a copy of that modified model and then I change the name of model, receiver # and I have to copy all the output values of the existing old model and enter them in this new copy....

This is a way of doing it by hand.
If the copy of outputs could be done on the radio, this would be of great help.

Another way could be achieved by pasting all the values from one model on the other beside the Name / Rxnumber / Failsafe / Outputs

projectkk2glider, 2015-08-17:
>I create a copy of that modified model and then I change the name of model, receiver # and I have to copy all the output values of the existing old model and enter them in this new copy....

Sorry I don't understand. If you make a copy of the model (either on the radion or in Companion) **all model settings** are duplicated including output. Why would you then need to copy them separately?

fraternl, 2015-08-17:
I don't want to copy all data....
I have 4 DLG's in my radio.
If I improve the setup of 1 of them I need to copy that to these other 3 models...

But if I do that, I need to change the name back to the original one/ the receiver# and all data on the outputs page.

I could do a copy of that new model and instead of pasting it to these other 3 slots I would do a "partial paste". I would not overwrite these specific settings... Otherwise I have to do all that by hand.

griebd, 2015-08-17:
just to make it more clear to me, you do have 3 models where the only difference is the output/servos configuration? if that is the case, the new feature/wish needed is the possibility share the config between models....

fraternl, 2015-08-17:
That would be a more complex thing to write.
It would also be difficult to manage...
A "partial paste" would keep the individual character of each model.

kilrah, 2015-08-17:
There is about a million of possible "partial pastes" though, impossible to know what exactly you want copied and what you don't (yes as usual your one particular case is pretty well defined, but if that gets added we'll have 20 people requesting different sets of data afterwards). Can't see a way to make something that makes sense.

How about you just replicate the things you changed instead of making a copy of the whole model everytime.

fraternl, 2015-08-17:
Most of the errors made are done in the logical switches / flight mode / special function combination. It also is almost impossible to do on the radio itself.

I wouldn't want "a million ways"... some models would still need a little handwork, like the 2S-setup instead of a 1S-setup. As long as I can be sure that things likes curves, flight modes and switches are an exact copy.

If you want a different setup then make a different setup.
It is intended to be used for similar models.

A partial paste that might need a little tweak

projectkk2glider, 2015-08-17:
Ok an idea:

* right click on source model and select `Copy`

* right click on the destination model and there you have:
  * existing `Paste` which copies the whole model (you don't want that)
  * several new Paste options, like `Paste mixes`, `Paste Logical Switches`, etc. echo one only copies specific part of the original model.

Would that satisfy you? Of course this would only be possible in the Companion, the radio can not do that.

kilrah, 2015-08-17:
To me the potential for errors that way is even higher than doing it manually. Paste mixes, some may depend on orphan LSs or Inputs that haven't been copied, etc...

Outputs is the only page that could be pasted as a whole IMO as it doesn't have dependencies (apart from the curves...)

projectkk2glider, 2015-08-17:
That is true. As always with power comes the danger and responsibility.

fraternl, 2015-08-17:
"Outputs is the only page that could be pasted as a whole IMO as it doesn't have dependencies (apart from the curves and the extended limits setting...)"

Copying outputs would be a possibility and would already be quite helpful.
In practice I would then first have to make a backup of the "child", then copy the "father" on the original child and then paste the "outputs" of the backup on the child....
I would then still need to alter the name, RX# and failsafe settings...

fraternl, 2015-08-23:
I would really like to have this feature to paste only 1 page onto another model.
Doing 1 page at a time would  already help me a lot and speed up the distribution from 1 model to the other 3.
Making sure that some pages keep their original values would be great.

In my case I would copy the setting of the model for which I did the latest changes...
Then I would paste the mixers page, the logical switches, the special functions...
Sometimes (after some rigorous changes) the flight modes, global vars as well.

projectkk2glider's proposal would work fine with me...

BTW...  are you guys on holiday? last week there was a lot less activity or am reporting too much?

fraternl, 2015-11-05:
Last weekend I wanted to replace the "logical switches" and "special functions" tab of a model with that of another model. Using Companion and the compare function this took me 2 hours.

It would have taken me 5 minutes with a "paste page" as was suggested by projectkk2glider.
I think it's an option many of us would appreciate if it is there.

kilrah, 2015-11-05:
It already takes 5 minutes, albeit with more clicks.

fraternl, 2015-11-22:
No, it can't be done in 5 minutes. Not with my setups.

kilrah, 2015-11-22:
Open both models, copy/paste each line from one to the other.

fraternl, 2015-11-23:
One has to do the telemetry page first otherwise you can't use them in logical switches. This will take 15 to 20 minutes by hand and you're not sure it's exactly the same. I'm using all the different cells and have 3 pages telemetry screen.
Then I have to copy all 32 logical switches which takes about an hour and then all special functions.
Even if you can do all this in half an hour it would only take you a fraction of that.

Is it you believe I'm slow you don't think it's worth this feature? 
I have a nice throttle cut feature and I don't want to risk any injury because I made a little mistake in copying the logical switches/special functions. Some other errors might make me lose a model.
Last new model I copied as a whole and patched it afterwards.

kilrah, 2015-11-23:
> Is it you believe I'm slow you don't think it's worth this feature? 

I believe you put more importance on this feature than it has, precisely because you say things like

> Then I have to copy all 32 logical switches which takes about an hour

Which makes no sense given you can easily copy&paste them. Yes you have to do up to 32 copy/paste operations, but on 2 windows open next to each other that only takes a couple of minutes.

fraternl, 2015-11-23:
To be honest, I never expected a copy paste to work across 2 windows, so up to my next try I did them all by hand.
One has to do the telemetry page before, though...

I will do a typical model next time using copy/paste and see how much time I lose then. It will become a repetitive action and you know I'm good at that... ;-)

But project2kglider seemed inclined to make something. Would there be any hope there or should I just let go and curse each time I have to fill these fields?

Thanks fit the copy/paste tip...

closed 2016-11-27 by fraternl