opentx/opentx #2722: Random Haptic Feedback

cpwilson, 2015-08-16:
Experiencing random haptic feedback on Taranis Plus on 2.1.1 firmware.

Only experienced the issue on a model where the internal radio is off and external module (immersion uhf) is enabled.    At random times there is haptic feedback pulses for no apparent reason.   

Also when selecting the model using uhf have seen the Control Warning check screen fail to display the switches and the display is corrupt until radio is power cycled.

cpwilson, 2015-08-16:
Reverted TX back to firmware 2.0.17 and issue is not present.

kilrah, 2015-08-17:
What ezuhf version? The JR module or "big one"? The big one is known to provoke strong fields that cause undesired operation if it's too close to some radios.

cpwilson, 2015-08-17:
It's the JR module version.    I am aware that the high power modules can cause RFI issues with the Taranis but this usually shows as the classic speaker buzz which have always experienced.    I did some more testing and I get random haptic only on 2.1.1 firmware.    Reverted back to 2.0.17 and all is well again as it has been for some time so it's not hardware related.

kilrah, 2015-08-17:
Did you reflash the same 2.1.1 file you had before? Due to your screen issue I wonder if it could be corrupted, might be worth redownloading. What exact filename are you flashing (so I know the options)?

cpwilson, 2015-08-17:
I have updated the firmware 2 times via companion and downloaded firmware file on each occasion and also tried via the bootloader on the TX itself but still have the same results.   Filename is opentx-taranisplus-en.bin.

kilrah, 2015-08-17:
OK.
Could you post your eeprom (upload to some file host and paste link here) and detail what firmware version and settings you're using on the EZUHF?
I don't see what could be the cause but I'll have access to a JR EZUHF in a few days and can try to reproduce.

cpwilson, 2015-08-17:
I'm running firmware V1.52 on the EZUHF module with stock settings apart from enabling 430-450MHz Extreme Hopping frequency mode.   Also running the Diamond SRH 771 antenna.  I have just checked and I did not take a full backup of the eeprom when running V2.1.1, only have individual model backups,  and have now reverted it all back to V2.0.17.     If it's of help to you I could drop 2.1.1 back on and copy the eeprom file off once done.

kilrah, 2015-08-17:
The individual backup of the affected model you have should be enough for now, I'll get back to you if I need more.

cpwilson, 2015-08-17:
bin file for the model is here:

https://drive.google.com/folderview?id=0B_3ok934_pP5fkpiU3FMcm9lLUFRbnMzdFdHM2s1eUdGVXN0dGREZkhMX1ZrcU5lYU9mUUE&usp=sharing

kilrah, 2015-08-18:
This model has the internal module ON in D16 mode and the external one OFF so I doubt it is the correct file...

cpwilson, 2015-08-18:
Hi Kilrah, it's showing correctly in companion and was definitely working on the TX when I did the backup.   Here is the screenshot of companion of the bin file.

I also tried creating a new model and on disabling the internal module and enabling the external module the random haptic started so it's nothing on the model setup.

[image]

bugobe, 2015-08-23:
I just purchased a Taranis Plus along with the EZUHF JR module today. I updated everything to the latest firmware. 2.1.1 on the Taranis and 1.52 on the EZUHF. I have the same problem with the random haptic pulses. I'll try downgrading the firmware and see if it makes a difference.

cpwilson, 2015-08-23:
Would be interesting to see if the problem goes away for you if you downgrade to firmware v2.0.17.
Keep us posted on your results.

bugobe, 2015-08-23:
After downgrading to v2.0.17 the problem is gone. I verified that the module is working and communicating with the RX in both fw versions.

kilrah, 2015-08-23:
OK, I got my EzUHF back, updated it to 1.52 and can't seem to reproduce.

Can you detail the conditions? low and/or high power? Does it happen every 5 seconds or once an hour? Does it happen the same whether the receiver is powered or not? @cpwilson when did you buy your module?

bugobe, 2015-08-23:
Both the Taranis and module are new as of yesterday. It happens on low
power for me only when the module is powered and haptic is turned on. If
haptic is set to "quite" it's fine. It pulses probably once every few
seconds. It almost made me think of Morse code when it first started.

kilrah, 2015-08-23:
OK. 

Do you have the latest 2.1 sound packs (Aug. 7)? When you do, do you have any sound playing with the vibration?

bugobe, 2015-08-23:
That I'm not sure of, all I did was install the 2.1.1 firmware and then
found the problem and then downgraded to 2.0.1 7. I will have to do a
little research on how to upgrade the voice pack but I will post back once
I've tested it.

kilrah, 2015-08-23:
OK. Then note that 2.0 and 2.1 need different sound packs.

bugobe, 2015-08-23:
Ok, I updated the voice pack to Amber v2.1 for the Taranis Plus. Now, with
the EZUHF on low power I can't get it to vibrate but as soon as I flip to
high it vibrates and then it randomly vibrates about 1-2 times every 30
seconds. It also vibrates once on the switch flip back to low power but
thats it once its on low power.

cpwilson, 2015-08-23:
My module is about 7 months old and is definitely running Version: 1.52 (6), just double checked.
It's been in my Taranis Plus, on the following firmware revisions with no problems at all.
FRSky V2.0.9.2
OpenTX V2.0.15
OpenTX V2.0.17

I updated to V2.1.1 firmware, as well as installing the sound pack and get the random haptic on both high and low power, but less frequent on low power.
As soon as I revert back to V2.0.17 and restore model all is fine again.

Having tried out all the new telemetry on 2.1.1 I would love to stay on it but can't as I need to run the EZUHF for some of my FPV models.

kilrah, 2015-08-23:
OK, I take it you have no sound at the same time than the vibration then.

bugobe, 2015-08-23:
Oh sorry forgot about that. No sound, you can hear a little static if you
hold the TX close to your ear.  But nothing coinciding with the haptic.

cpwilson, 2015-08-24:
Same here, no sound just the vibrate and not using haptic for anything in the model setup.

johnho100, 2015-09-01:
Hi! I am having the same problem after updating to 2.1.1 firmware, I am getting random haptic feedbacks when I turn on my Ezuhf JR module. I also tested an Openlrs JR module, same issue. The haptic feedbacks happen randomly when no buttons or sticks are moved. I am able to repeatably "trigger" an abnormal haptic feedback when I press the ENT button (happens only if the LED backlight is off). 

I flashed back to 2.1.0 firmware, the problem disappeared. It seems the issue is caused by changes in 2.1.1 firmware only.

GRCT42, 2015-09-09:
I have the same problem as you guys. This is very frustrating so i just set it to quiet mode. Hopefully it will be resolved in a Firmware fix!!

bsongis, 2015-09-09:
@opentx/commiters Who has this module? We need to do the dichotomy and find which is the wrong commit!

bsongis, 2015-09-09:
Unless anybody here is ready to test firmwares I would send to him. If yes please contact me by email.

GRCT42, 2015-09-09:
I have the JR Module in a Taranis Plus. I have saved a copy of my Radio .Bin file if you need it.

I have the same problem as (bugobe )"Ok, I updated the voice pack to Amber v2.1 for the Taranis Plus. Now, with the EZUHF on low power I can't get it to vibrate but as soon as I flip to
high it vibrates and then it randomly vibrates about 1-2 times every 30
seconds. It also vibrates once on the switch flip back to low power but
that's it once its on low power."

Any help appreciated.

kilrah, 2015-09-09:
> @opentx/commiters Who has this module? 

I have one, but unable to reproduce...

GRCT42, 2015-09-09:
AsI stated in the forum, I have the JR Module in a Taranis Plus. I have saved a copy of my Radio .Bin file if you need it.I have the same problem as (bugobe )"Ok, I updated the voice pack to Amber v2.1 for the Taranis Plus. Now, with the EZUHF on low power I can't get it to vibrate but as soon as I flip to
high it vibrates and then it randomly vibrates about 1-2 times every 30
seconds. It also vibrates once on the switch flip back to low power but
that's it once its on low power."Any help appreciated.
 K. Pike

bsongis, 2015-09-09:
Would it be possible that a signal comes to the module (heartbit PIN) and OpenTX thinks it is a trainer/trainee connection? In this case it would bzzz since we added this bzzz in 2.1.1

kilrah, 2015-09-09:
I thought about that but then the trainer found/lost sounds would play (checked they had the latest pack). 

But now someone made an interesting suggestion, the UHF could send noise into the analog path and cause the "center beep" to trigger. 

Can you look in the analog diag page if you have a big difference in "jumpiness" of the values with the UHF module on and off, and report what settings you have for the center beep?

bsongis, 2015-09-09:
Interesting, the user could try to "de-calibrate its sticks" to see if the bzz is done

cpwilson, 2015-09-12:
Hi, My module is the EZUHF.

I have tried de calibrating the sticks by doing a reset and not going through the calibration procedure and I can confirm the random haptic is still present on 2.1.1 and 2.1.3 firmware versions.
When I set haptic to Quiet then the issues goes away but when it's enabled random haptic is back.

Have also disabled the external module and there is no noticeable difference with the fluctuations in the channel monitor page with the module on or off.

I get random haptic when module is in the high or low power setting but less frequent in the low power setting.

johnho100, 2015-09-17:
Hi, I confirm I am still experiencing the same issue after flashing to 2.1.3 , when I switched back to 2.1.0 , the problem disappeared.

flythai, 2015-09-21:
I, i have the same problem, i no have previous versions open TX firmware, and i ask if i disable the haptic function have any problem.? Many thanks.

er9xuser, 2015-09-22:
Hi. I've the same problem (random vibrations) on Taranis Plus and openTx v2.1.3 but I'm using RLink UHF connected to pins for JR module.

bsongis, 2015-10-13:
Did someone check with our latest nightly builds? We have found 2 major bugs recently. It would be nice it is a side effect...

cpwilson, 2015-10-13:
HI, I have just tried 2.1.4N488 build on my Taranis Plus with the EZUHF module and unfortunately the Random Haptic is still present.   As such I'm sadly holding off upgrading from 2.0.17 until a fix is sorted as I use the EZHUF on my main model.

cpwilson, 2015-10-13:
I may be wrong but the random haptic does not seem as bad, less often, as it was in 2.1.3.

projectkk2glider, 2015-10-19:
Lets get some additional information:

Affected users please do this test:
* as reported when Haptic is set to `Quiet` this issue does not appear.
* set Haptic to `Alarm`, does the issue appear? If not then continue.
* set Haptic to `Nokey`, does the issue appear? If not then continue.
* set Haptic to `All`, does the issue appear?

More questions for affected users:
* what is the pattern of haptic, only one short vibration or several short vibration pulses?
* what is your setting for `Beep mode`?
* do you have files `trainko.wav` and `trainok.wav` on the SD card in `SOUNDS/en/SYSTEM`?

My bet is `AUDIO_TRAINER_LOST()` and `AUDIO_TRAINER_BACK()` because at least Kilrah's sound pack for 2.1 does not contain `trainko.wav` and `trainok.wav` files.

We also saw in #2913 that the trainer input detection is not working correctly (reported to be circuit problem) and it might be floating and affected by high RF levels.

flythai, 2015-10-19:
For me only in quiet mode no have vibrations, another all modes i have vibrations.
If i put haptic to alarm-nokey-or all, sometimes i have 1 short vibration and another times i have 2 sorts vibrations.
I no have the files trainko on the sdcard.
My settings for Beep mode is Nokey.

projectkk2glider, 2015-10-19:
@flythai please create trainko.wav and trainok.wav files (use any other file, copy, rename) and tell us if it plays when you get the random haptic vibration.

flythai, 2015-10-19:
Give-me 30 minutes and i tell you, any wav file, i rename.

flythai, 2015-10-19:
I again, i just make the experience, before the experience i just have haptic vibrations when i put the jr uhf on HI mode.
I rename the 2 files, and when i put ALARME and the jr module on HI mode, i have the same vibrations with sound wav.

projectkk2glider, 2015-10-19:
Thanks, that confirms it.

flythai, 2015-10-19:
This is have any solution ?
Or if i Put on Quiet mode i no have problems ?

projectkk2glider, 2015-10-19:
Now that we determined where the problem is, we will prepare a fix for it.  You  will be able to see the progress if you follow this issue.

flythai, 2015-10-19:
I follow this from begining, i hope you guys can fix this problem, i pray.

kilrah, 2015-10-19:
Meh, I asked 2 months ago of they were using the latest and full sound packs because it's the first thing that I thought was causing it. Answer was yes...

ImmersionRC have apparently changed something on their module though, because mine does not put anything on the affected pin.

bsongis, 2015-10-19:
I guess it's only an '''else''' missing in trainer_driver.cpp

bsongis, 2015-10-25:
Latest nightly builds now include the fix (I launched the last one this morning). I will need tests before closing the issue.

flythai, 2015-10-25:
Great bsongis, i hope result your tests

bsongis, 2015-10-25:
Did you try? I don't have the module myself ...

flythai, 2015-10-25:
I no see the new firmware update to try

bsongis, 2015-10-25:
You have to download Companion from nightly builds here:
http://downloads-21.open-tx.org/nightly/companion/
Then you enable firmware nightly builds in Preferences and download the latest nightly built firmware.

flythai, 2015-10-25:
The version of companion is 2.1.5n504.exe Correct ?
And the firmware version is -i686.rpm correct ?

projectkk2glider, 2015-10-25:
>The version of companion is 2.1.5n504.exe Correct ?

Correct

>The version of companion is 2.1.5n504.exe Correct ?

No that is Companion for RedHat based Linux distributions. You will download the radio firmware with the Companion2.1.5n504.exe when you install it and then make sure you enable `firmware nightly builds` in Preferences and then download the latest nightly built firmware.

flythai, 2015-10-25:
Inside companion where i can download the firmware nightly builds ?
I just check on firmware nightly builds on preferences.

[36 more replies]

commit 95040234 (2015-10-25): Another attempt to fix #2722
commit 942148fc (2015-10-26): Another attempt to fix #2722
closed 2015-10-26 by bsongis