opentx/opentx #2725: X9E rotary pot extreme clockwise causes value to count down

timrowledge, 2015-08-17:
Hope the title conveys the basic issue ok.
To reiterate a posting on rezone - 
"I found a curious issue when calibrating my sticks & pots yesterday; it's mainly a slightly-faulty pot issue but has an interesting software component.

The right pot has a slight 'graunch' to it at the extreme clockwise end of travel. I noticed but dismissed it as perhaps a bit of swarf needing cleaning out someday. However, when calibrating I saw that the reading went up to 100, and then started counting down! So far as I can make out the reading is fine up to where the scrappy noise occurs and then that last tiny bit of rotation causes the value to commence a countdown. Not a sudden drop to a different value, an actual 99-98-97-96 etc count. I'm pretty sure that isn't what is intended.

The hardware is no big deal, I'm sure Aloft will sort that out lickety-splick but I can imagine some very strange in-air behaviour if it happened in flight. And I certainly have trouble imagining what corner of the software could cause a countdown of a pot reading. If it's something that could be triggered on a main stick pot it might be more than an amusing oddity."
I haven't been able to come up with anything more to narrow it down. It's just the one pot causing it. Perhaps the countdown effect is a side effect of how the ADC works - I've been having plenty of fun with those myself as I prepare software for the Raspberry Pi  ISS mission.

kilrah, 2015-08-17:
99% it's a faulty pot, and the "count down" is simply due to the pot putting out a voltage that falls again where it's faulty... or just leaving the wiper floating. Nothing we could do about that.

timrowledge, 2015-08-17:
I'm pretty sure it's a faulty pot and that can just be replaced. No argument there. The bit that interests me is the steady count down; that's the sort of pattern that has so often in the past lead me to a subtle but alarming bug.
Since my life is already far too full of development projects the best thing I can think of would be to mail someone the faulty pot and see if it triggers anything for an actual opentx developer. At least having it recorded means anyone else seeing such behaviour in the future has a hook to hang it on.

projectkk2glider, 2015-08-17:
@timrowledge can you measure the pot? I would be interesting to see the wiper resistance in that region.

kilrah, 2015-08-17:
Yep. If the pot fails open a rather steady countdown would be totally expected due to the input cap slowly discharging itself.

projectkk2glider, 2015-08-17:
@kilrah you might be wrong there. Remember we read many other analog signals with the same ADC input, so the cap gets recharged many times before we get to the faulty pot.

kilrah, 2015-08-17:
I mean the external cap that is on the board between the pot and CPU, not the CPU's S&H. This will initially be at pot voltage then get discharged a little everytime it's sampled and will tend to reach the previously sampled input after a while.

projectkk2glider, 2015-08-17:
You are right! I forgot about the 100nF cap. It is definitely big enough to cause such behavior.

timrowledge, 2015-08-17:
After pulling the pot to plug in the ol'meter it is clearly faulty; in fact I can now rotate it 360deg. When going past the 'proper' end point it goes open circuit, so it seems to me we have a solution that makes sense. I'm glad the ADS1015 adc I'm using doesn't do the slow fade thing!

kilrah, 2015-08-17:
It would do exactly the same if it was hooked to the same circuit.

closed 2015-08-17 by kilrah