opentx/opentx #2726: Include version number in boot screen

pinkywafer, 2015-08-17:
It may be helpful to have the major version number included in the default splash screen ("OPEN-TX"), to improve the recognition of which version of users are running:

ie. "OPEN-TX 2.0" on 2.0.18......
"OPEN-TX 2.1" on 2.1.2.....

Just a thought
Jenny

projectkk2glider, 2015-08-17:
Difficult to implement cleanly. The splash screen graphics is actually a graphics and is also user changeable. So writing any text (like version) over it will perhaps overwrite some (important) part of the graphics. 

Although it might not be very intrusive if the text was in one of the corners. But then again the splash screen can be turned off by the user.

pinkywafer, 2015-08-17:
I appreciate what you mean, but.. I had just meant by editing the png file for the default screen - it may prove helpful for the less "experienced" users, who may not be changing the splash screen.

something like 
[image]

If the user replaces it, they're more likely (I believe) to have more of an idea what they're doing!

projectkk2glider, 2015-08-17:
Ok, I have changed the title of the issue to better reflect your idea. Lets see what other devs say.

pinkywafer, 2015-08-17:
OK, thanks

kilrah, 2015-08-17:
Why not. Although I'm afraid it might not be that useful as I've seen replacing the splash screen is one of the first things even the less knowledgeable users will do...

rguptaca, 2015-10-04:
How about maybe adding a line at the top of the model selection menu? We cisit that menu all the time? Also boot loader version maybe.

closed 2019-11-15 by stale[bot]