opentx/opentx #2728: Edge {boolean} [00:--]  --- 2.0   becomes true upon selecting a model

fraternl, 2015-08-18:
I created a switch that has to be true 2 seconds after leaving FM1

Edge FM1 [00:--] --- 2.0

That switch will put my model in FM4
FM4 triggers "Play Track launch"

When I choose my model it will go into "launch" (FM4) and play the track "launch"

I tested other input values instead of FM1 (ones that are for certain false) and they also make this switch become true for 2 seconds.
Further behaviour is as expected...

LapinFou, 2015-08-18:
you forgot to specify which version of OpenTX you are using...
;-)

tomkfly, 2015-08-18:
If you mean 2.1.x nightly builds then nothing has changed, as it is the same in 2.0.17.

fraternl, 2015-08-18:
It has probably been that way all the time.
Can anyone confirm/deny the issue?

projectkk2glider, 2015-08-18:
I will take a look..

projectkk2glider, 2015-08-18:
Yes it a subtle but, which is been there for a long time. I will fix it.

fraternl, 2015-08-18:
Mmmm...
Is this for the time being and might it get a fix some day?

bsongis, 2015-08-18:
@fratern what do you mean?

fraternl, 2015-08-18:
Hi Bertrand...
I misunderstood the fix...
I noticed some text and thought this would not get a fix, but merely a warning was given that this could happen.
I then saw some more stuff and noticed it's being worked on at this moment..

projectkk2glider, 2015-08-25:
I will reopen this in order to remember to port this to 2.0 branch

projectkk2glider, 2015-09-20:
Ported to 2.0 in https://github.com/opentx/opentx/commit/f9946dff032f9e4a8bbe1b29535b41459dce162e

closed 2015-09-20 by projectkk2glider

fraternl, 2016-01-08:
I checked these versions

2.1.0,  2.1.5 and today's nightly

I have this setup

L1 ( Edge(SE-, [0:instant]) ) AND SH↑ Duration(1s)
L2  Edge(SE↓, [0:0]) Duration(2s)

When I turn on the radio the switched are set as   SE-   and SH↑

I now noticed there are actually 3 bugs.
All of them are fixed in latest nightly

-- V1 - bug   the boolean in V1 should not influence the Edge behaviour, merely masking end result
-- Upon selecting model a false edge value triggers
-- Upon selecting model a true edge value does NOT trigger

2.1.1    good, bad,  bad
2.1.5    bad, good, good
nightly  good, good, good

I did not test the 2.0 version on the radio, but I think all of them (except 2.0.18) are like 2.1.1
2.0.18 behaves like 2.1.5

I hope a 2.0.19 gets released that will act like today's nightly.

@projectkk2glider   can you confirm that this 3rd situation is now acting correctly (upon selecting a model with the edge in true it should be triggered)?

I wish https://github.com/opentx/opentx/issues/2837 was solved different.
I prefer to have everything the same....

commit 537d7e08 (2015-08-25): Fixes #2728: edge switch could trigger right after the logicalSwitchesReset() was called
commit 5e2fe112 (2015-08-25): Merge pull request #2729 from opentx/projectkk2glider/issue_2728_edge_bug

Fixes #2728: edge switch could trigger right after the logicalSwitche…
commit f9946dff (2015-09-20): Fixes #2728: edge switch could trigger right after the logicalSwitchesReset() was called (ported from 2.1)