opentx/opentx #2733: Minor tweaks to Dutch Translation

fraternl, 2015-08-19:
http://hmvc.eu/nl.h.txt

projectkk2glider, 2015-08-21:
Hi, your translations always change `TR_VCSWFUNC` namely the character `Δ` in your file is `Î`. Is this your intention?

LEN_VFSWRESET should remain \012 (octal) which is 10 (decimal) as `#define TR_FSW_RESET_TELEM   TR("Telm", "Telemetrie")` is 10 chars long.

Other changes are OK.

closed 2015-08-21 by projectkk2glider

fraternl, 2015-08-23:
Nothing intentional there....  I will check this. I'm using a comparison tool... I will focus on these changes...

fraternl, 2015-08-23:
I'm using en.h.txt as a reference.
Examdiff can't detect a difference between nl.h.txt and en.h.txt on those lines

#if defined(CPUARM)
#define TR_VCSWFUNC            "---\0 " TR_CSWEQUAL "a~x\0 ""a>x\0 ""a<x\0 " TR_CSWRANGE "|a|>x""|a|<x""AND\0 ""OR\0  ""XOR\0 " TR_CSWSTAY "a=b\0 ""a>b\0 ""a<b\0 ""Δ}x\0 ""|Δ|}x" TR_CSWTIMER TR_CSWSTICKY
#else
#define TR_VCSWFUNC            "---\0 " TR_CSWEQUAL "a~x\0 ""a>x\0 ""a<x\0 " TR_CSWRANGE "|a|>x""|a|<x""AND\0 ""OR\0  ""XOR\0 " TR_CSWSTAY "a=b\0 ""a>b\0 ""a<b\0 ""^}x\0 ""|^|}x" TR_CSWTIMER TR_CSWSTICKY
#endif

fraternl, 2015-08-23:
After changing \011 to \012 the file nl.h.txt on github is identical to the one I have here online....
strange....

projectkk2glider, 2015-08-23:
Strange indeed. And it looks like your diff tool completely ignores the utf8 encoded delta character. Maybe your tool can not handle UTF-8 encoding.

commit 60c925a7 (2015-08-21): Fixes #2733: Minor tweaks to Dutch Translation