opentx/opentx #2734: [Companion 2.1.1] Companion crashed when using "Compare Models"

LapinFou, 2015-08-19:
Step to reproduce the problem:
1- Download the EEPE <a href="http://1drv.ms/1JhF1QV">nano-cpx_LapinFou.eepe</a>
2- Drag and drop the same NANO_CPX model on the right and left side.
3- Companion has crashed.
Environment: Windows 7 Pro 64 bits.

projectkk2glider, 2015-08-19:
Which radio is selected in the radio profile?

LapinFou, 2015-08-19:
Taranis Plus
[image]

closed 2015-08-19 by projectkk2glider

fraternl, 2015-08-20:
What I was hoping for came true....
This change now fixed the curves....
They are now showing

LapinFou, 2015-08-20:
Nice !!
:)

fraternl, 2015-08-20:
BTW... it's only partially fixed...

projectkk2glider, 2015-08-20:
>BTW... it's only partially fixed...

What do you mean? Are you still having problems, details please.

commit 3075ad85 (2015-08-19): Fixes #2734: fixed endless loop in model compare (varable i was used both in outer and inner for loops)
commit f481d2f2 (2015-08-19): Fixes #2734: fixed endless loop in model compare (varable i was used both in outer and inner for loops) (ported from 2.1)