opentx/opentx #2737: Wish: Separate curve graphs in "Print Model" and "Compare Models"

fraternl, 2015-08-20:
In "Print model" and "Compare models" all the curve graphs are merged together in 1 image.
These overlapping curves soon become unreadable if more than 2 are shown

It would be nice to have a graph for each curve

[image]

bsongis, 2015-08-20:
Please nobody works on Compare and PrintDialog, I am on it

projectkk2glider, 2015-08-20:
Here are related issues to this one:
*  Companion print model and compare model has receiver number missing #2576 
*  [OpenTX2.1] [Companion] Compare Models #1862 
*  Companion Print model needed enhancements #1063

commit 0f91d66b (2015-08-23): Merge pull request #2752 from opentx/bsongis/comparedialog_printdialog_refactoring

PrintDialog / CompareDialog refactoring. Fixes #2737 and #2576. I also completes some of the tasks in #1063.
closed 2015-08-23 by projectkk2glider