opentx/opentx #2739: [Enhancement] Random generator

LapinFou, 2015-08-20:
Hi folks,

I saw a nice B17 video, where the guy created a random generator on his Spektrum DX18 using 2 mixes. He is using it, to randomly move the gun turrets on his plane.
Could be a nice feature for people flying with scale models.
:)

fraternl, 2015-08-20:
We can't let them have a feature that's not in OpenTX, can we?   ;-)

kilrah, 2015-08-20:
That would be the perfect job for a Lua script.

LapinFou, 2015-08-21:
@fraternl : Absolutely ! We can't... ;-)
@kilrah : Yes, but I didn't see a random  function in LUA (I checked before opening a ticket).

kilrah, 2015-08-21:
Isn't there a math.random()?

LapinFou, 2015-08-21:
My mistake, I checked the OpenTX doc, but forgot to check the official LUA libs...  :-/
Well, I guess this ticket can be closed.

mhotar, 2015-08-22:
It could be nice feature to have special function that can set Gvar to random value. I think it is more valuable to have it in the base function set.

kilrah, 2015-08-22:
I would think any use of this would need a bit more processing to be really useful...

mhotar, 2015-08-22:
It could be processed through standard mixer path.

closed 2019-11-15 by stale[bot]