opentx/opentx #2740: SWR Value at X9D and X9E  changes

hrenz, 2015-08-21:
Hy,

SWR Value
Normal is between 0 to 1 thats ok
if I take may Hand round the antenna SWR rises  form 15 to 18 that ok
if I take may Hand away from Antenne SWR will stay at last value 15 to 18
thats not ok, must be fall back to 0 to 1

this effekt is  much stronger  at X9E because there is the handle

if i do this often shortly after the other I can generate and store lots of different SWR "Values"
sometimes 0 and 1 too.

think ist only a dirty effekt from FrSky SWR-Signal Output to openTx

This behavior makes many people uncertain

janekx, 2015-08-21:
2.1.2N425 Taranis-E
SWR start show the value when the RX is connected, before it not show any number and after disconnecting RX stop show - freeze at last number. 

The hand test works well covering goes to 17 and relasing drop down to 0-1 in a second or two. 

Cant confirm this issue, there is space to discuss to show SWR even no RX connected but no any issue at SWR display on X9E

hrenz, 2015-08-21:
Hy,
X9E is connected to a X8R, Telemetry is running
SWR at start is fine and between 0 to 1
The hand test works well covering goes SWR to 17 but don't relasing drop down to 0-1  
not in 1 or 2  sec, not in 1 min
With hand test I can generate lot of different SWR Values, sometimes 0-1  with the hand
Take the X9E  at the handle SWR rises up, but don't go back thereafter.

I can show a video if you it

kilrah, 2015-08-21:
That has nothing to do with openTx.

closed 2015-08-21 by kilrah