opentx/opentx #2741: Fix trim step to ‘Coarse’ when used as source

pindeslandes, 2015-08-22:
Taranis - OpenTx 2.0.17
When you use a trim as source in a mixer (or input), the problem is that the sensitivity of your control is linked to the trim step you have set in the general settings of your model. So if you change your trim precision for whatever reason, you change your control sensitivity, which you generally didn’t intend to.
So maybe it would be a good thing that the trim step you set for trimming purpose had no influence on the sensitivity of the trim used as source.
Fixing the sensitivity of the trim used as source to ‘Coarse’ might be a good value, as it would give 32 steps from -100 to 100 (the equivalent of the throttle ratchet).

kilrah, 2015-08-22:
Definitely not the way to do it, expecting everybody to be happy with Coarse when using the trim as source is no good at all. I'm not in several of my models...

pindeslandes, 2015-08-22:
Or maybe an independant choice for trimming and source?
Trim as source sensitivity could be choosen between:
- Coarse
- Medium
- Fine
- Extra fine
- Exponential
- Same as Trimming
With the default beeing 'Coarse' or 'Same as trimming'?

Sacre100, 2015-08-22:
Very similar to this one #1814 or #1824 already rejected.

closed 2019-11-15 by stale[bot]