opentx/opentx #2743: Companion sync SD-card does not overwrite older files

fraternl, 2015-08-22:
I don't know if it's supposed to do, but the sync option does not overwrite older files.
I noticed that it only transferred 1 csv-logfile although I flew with 3 models today.
I had already 2 of today's log-files on my PC.
They were from before today's flying session

fraternl, 2015-09-03:
But is it not implemented or is it failing?
It thought it already was 2-way, but I'm not near my radio to check...

projectkk2glider, 2015-09-03:
The sync process check modification time for each file and if the existing file is newer than the one being copied, then it does not overwrite it. 

Can you provide an example of not working state and a listing (with times) of file in both directories?

fraternl, 2015-09-03:
Sorry... I'm now mixing issues...
AFAIK it is 2-way and never noticed any problems with it....
Someone else made a ticket #2824 where he requested a 2-way sync....

My ticket (this one) is about the sync process being able to update a file if it is newer.
It seems it is not implemented.
Will it one day?

kilrah, 2015-09-03:
@projectkk2glider 's post suggests it should already do.

fraternl, 2015-09-03:
I just read it again... Yes, you're right...
Tomorrow evening I will do some tests as I don't have my radio during the week.

I noticed it because I already had a logfile of that day on my computer and it therefore left that old copy intact although there was already a newer one on my radio.

Maybe someone else can check it before me.

bsongis, 2015-09-03:
Yes the SD sync process is 2-way and overwrites older files, unless you
found a bug.

bsongis, 2015-09-03:
Duplicate of #2824

closed 2015-09-03 by bsongis

fraternl, 2015-09-03:
That other ticket was made later and has only 1 sentence.
Now my ticket gets closed. Strange...

kilrah, 2015-09-03:
Because there are 2 about the same thing and the comment about the more likely reason is in the other one. And you mixed up things in this one too making it a mess. Give us a break.

bsongis, 2015-09-03:
We need only one issue for this problem and the other had more interesting info (the Logs and Backup directories)