opentx/opentx #2746: Trigger haptic with special function in any haptic mode

pindeslandes, 2015-08-23:
Taranis – OpenTx 2.0.17
The radio general settings define the general circumstances when the haptic should be automatically allowed or not (haptic modes), which is fine.
But the ‘Quiet’ mode, for instance, totally disables the haptic even in haptic special function, where the circumstances are supposed to be mastered by the user and are linked to each model. And the user, may wish to trigger the haptic with the haptic special function whatever the haptic mode (i.e. even in ‘Quiet’ or ‘Only Alarm’ mode).
Typical use: turning a pot into a multi position switch, with haptic triggered each time you reach a new position, so that you can feel it under your fingers. No need to be related to the haptic mode here, in my opinion.

projectkk2glider, 2015-08-23:
Well both ways are correct, the way it is now and your way. If we change it then some other user will be unhappy. I am against the change, lets wait what others have to say.

kilrah, 2015-08-23:
I'm against as well for coherence. The point of quiet is, just like for sound, to be sure not to have any output from the device.

pindeslandes, 2015-08-24:
The question is: considering the special function is supposed to be mastered by the user, isn’t that enough to be sure not to have output if you don’t want any?

Other suggestion which would make every user happy, I guess: what about a check box for an ‘Enable haptic special function in any haptic mode’?
Could be done as well for ‘Enable sound special functions in any sound mode’.

kilrah, 2015-08-24:
> The question is: considering the special function is supposed to be mastered by the user, isn’t that enough to be sure not to have output if you don’t want any?

If you see it the other way it means if I want to temporarily mute haptic or sound I'd have to delete all my SFs instead of just setting the global setting off...

pindeslandes, 2015-08-24:
Yes. That’s why I answered to myself with a more versatile proposal (check box to enable special function in any mode).

kilrah, 2015-08-24:
Yeah but just that would be kind of a kludge, I'd rather rework the whole system and use separate checkboxes for all event types and solve a couple of other issues at the same time.

closed 2019-11-15 by stale[bot]