opentx/opentx #2747: Voice alarms and warnings screens when exiting from model settings.

pindeslandes, 2015-08-23:
Taranis - OpenTx 2.0.17
When you exit from the model settings, where you may have moved some switches, you are presumably ready to fly. So it would be useful if the warning screens where displayed and the voice alarms where heard.

projectkk2glider, 2015-08-23:
Could be useful. It is closely related to the #2748 and should be done simultaneously if we decide to implement it.

kilrah, 2015-08-23:
Dangerous and not good at all, the warning screens are designed to stop RF transmission until everything is cleared. It's OK when you change model as when you do that your receiver is supposed to be off, but going in (and exiting) the menus with the receiver powered is normal. You'll be triggering failsafe everytime you exit menus, if it's not properly configured...
Not to mention it's not unusual to fly while in the menu when doing adjustments.

projectkk2glider, 2015-08-23:
I agree with Kilrah.

pindeslandes, 2015-08-24:
Maybe we have a safety issue here because if I push Kilrah’s logic, I should cycle power in flight to hear warnings after making adjustments! Just kidding because I wouldn’t advise anybody to make adjustments in flight thru the settings menu. Way too dangerous , IMO.

**Seriously, my point is: where is the need to stop RF transmission?**
Because:
1st case: you change model. In that case your Rx is off, so there is no danger in transmitting.
2nd case: you change settings. In that case you Rx may be on and it’s better to keep transmitting.

Besides, a situation where sometimes the warnings appear automatically (when changing model) and some other times, you must keep in mind to do something specific to get the warnings (cycling after setting) is not very safe, I think.

kilrah, 2015-08-24:
> if I push Kilrah’s logic, I should cycle power in flight to hear warnings after making adjustments!

You obviously wouldn't change a switch selection (so your case doesn't apply), but changing a weight or similar I've done many times.

> Seriously, my point is: where is the need to stop RF transmission?
For cases when people turn the RX on before the radio.

Frankly, in your case I'd just check my switches just like I do before every flight, the warnings are a safety net that like all such things you should never need, i.e. you should not use them as a way to know whether you're ready to fly.

And if you need the warnings to remember how the switches should be you've got another problem.

pindeslandes, 2015-08-24:
Well, I must admit I do use the switch warning screen as an easy reminder and that I find that very convenient!
It’s also very handy to set pots precisely before flight, regardless of safety matters. And I am afraid there’s no other simple way to do that.
That’s why I believe my request might be appreciated by many users.

So, maybe a solution to get those ‘warnings’ after setting could consist in stopping the transmission only if you change the model but not when you just change settings?

kilrah, 2015-08-24:
Yes that would be good... BUT unfortunately the 2 functions are currently tightly linked, and separating them would require extensive changes.

closed 2016-01-03 by kilrah