opentx/opentx #2748: Auto return to main view after model selection

pindeslandes, 2015-08-23:
Taranis – OpenTx 2.0.17
After changing model from the Model Selection page, and although you immediately get the relevant warning screens and voice alarms, you are still on the Model Selection page.
That is most of the time useless (generally, you just want to fly right after, not to go on with the model settings) and not very convenient (nor elegant either) because you need to press Exit to get back to the main view of the new model.

Maybe it would be nice if it could automatically go to the main view of the model.

Or, other suggestion:
When on ‘Select model’:
- ENTER short => go back to main view (easy access for ‘every day’ use).
- ENTER long => stays on Model Selection page (and shift the warnings screens and voice alarms) to go on with model settings.

projectkk2glider, 2015-08-23:
I like this idea.

fraternl, 2015-08-24:
The "enter long" shouldn't skip the warning screens as they also prevent the radio from transmitting.
The warning screen is intended to prevent potential dangerous situations like a motor that starts running or a rocket firing.

Having these warning screens you might have developed a sense of security. If it then suddenly bypasses that you might be in for a surprise.

I like to have this option too, albeit with one that never bypasses the warning screen.

projectkk2glider, 2015-08-24:
@fraternl Kilrah already explained https://github.com/opentx/opentx/issues/2747#issuecomment-133918147 that warnings are not possible in this case.

fraternl, 2015-08-24:
I don't have my radio here and I'm now not 100% sure that I know how it's working now...

closed 2020-07-07 by stale[bot]