opentx/opentx #2749: Mega 2560 stuck in bootloader when frsky telemetry

tmrttmrt, 2015-08-23:
Mega 2560 with cn340 does not boot when frsky telemetry is feeding data. Probably higher no. Tx/Rx ports should be used for telemetry to prevent the bootloader detecting serial port activity. I suggest swapping the rotary encoder pushbutton and telemetry pins to avoid this issue.

JivaroFAD, 2015-08-23:
This issue occurs only when receiver is running before turning on transmitter.
It can only be solved by a hard solution, with a single line driver (like SN74LVC1G34) activated by the "hold power" pin after boot. I have to do some test and will soon publish an updated scheme with this mod.

tmrttmrt, 2015-08-23:
I know, and do not like it. I would prefer just swapping the pinout since it requires a minimal hardware mod. Is there any particular reason to use tx0/rx0 for telemetry? Using pins 0 and 1 for pushbuttons would not be foolproof, but at least they are opened by default.

JivaroFAD, 2015-08-23:
Connecting 4 wires to a simple chip is much less work for me then make software modification that I don't know if it can work or not (TX/RX line is dedicated to this, I'm not sure other pins can work like theses pins...).
But if you think it is possible and have the time to write the code, I can push it on github.

tmrttmrt, 2015-08-23:
Maybe fixing the boot loader would be even simpler. I will do some research to see.

tmrttmrt, 2015-08-23:
I took a standard Arduino loader and based on this patch (https://github.com/arduino/Arduino/issues/2101) created a patch:

```
--- stk500boot.c.old	2015-05-06 17:46:34.000000000 +0200
+++ stk500boot.c	2015-08-23 16:31:50.132687300 +0200
@@ -272,6 +272,7 @@
 	#define BOOTSIZE 2048
 #endif
 
+#define MAX_ERR_COUNT       10
 #define APP_END  (FLASHEND -(2*BOOTSIZE) + 1)
 
 /*
@@ -541,7 +542,8 @@
 	unsigned char	msgBuffer[285];
 	unsigned char	c, *p;
 	unsigned char   isLeave = 0;
-
+	unsigned char   errCnt = 0;
+	
 	unsigned long	boot_timeout;
 	unsigned long	boot_timer;
 	unsigned int	boot_state;
@@ -713,6 +715,16 @@
 							msgParseState	=	ST_GET_SEQ_NUM;
 							checksum		=	MESSAGE_START^0;
 						}
+                        else
+                        {
+                            errCnt++;
+                            if(errCnt > MAX_ERR_COUNT)
+                            {
+								msgBuffer[0]=CMD_LEAVE_PROGMODE_ISP;
+								msgParseState=ST_PROCESS;
+                            }
+                        }
+						
 						break;
 
 					case ST_GET_SEQ_NUM:
```

tmrttmrt, 2015-09-03:
Should I put this into the source tree and create a pull request?

JivaroFAD, 2015-09-03:
I think it should be a great idea to put he source file of this bootloard in /radio/src/targets/mega2560

Many thanks for this great works, it's a much better idea than a hardware add-on like I was intending to do ... !

JivaroFAD, 2015-10-21:
Bootloader testing ok, thanks tmrttmrt :)
Source file added to /radio/src/targets/mega2560 (018a022c35a2ee7d22474363a27b43054636c9a6)

closed 2015-10-21 by JivaroFAD