opentx/opentx #2750: Mega 2560 servo movement not continous

tmrttmrt, 2015-08-23:
I observe a step-like servos movement when slowly moving sticks with frsky dht module.  It appears either like ppm output update frequency is too low or like the output resolution is truncated, since the movement of the sticks on the display indicators appears continous.

kilrah, 2015-08-23:
Or your servo has low resolution.

tmrttmrt, 2015-08-23:
It works normaly with another transmitter (Futaba fc-16 with frsky dht module) so it is not servo issue.

projectkk2glider, 2015-08-23:
There is no way of knowing which part is to blame. To find out you have to preform on of the tests:
* use OpenTX on mega 2560 with a different TX module (other model) if issue remains then OpenTX and/or mega2560 is to blame
* use existing TX module in another radio, if the issue remains then the TX module is to blame.

tmrttmrt, 2015-08-23:
I agree. I will do some more dignostic soon.

JivaroFAD, 2015-08-23:
Please check also the parameters of PPM modulation.

tmrttmrt, 2015-08-23:
I have played a bit with this already with no effect.

bsongis, 2015-08-23:
Did someone check the pulses interrupt latency?

tmrttmrt, 2015-08-23:
I have checked the existing module with the other radio and it works smoothly. I have checked also whether it is a static resolution problem by linking a channel with a full travel potentiometer and carefully slowly rotating it. From the servo deflection it does not look like a resolution problem, but rather a noncontinuous update of the servo position during the control input change.

How can I check the pulses interrupt latency?

kilrah, 2015-08-23:
Do you have access to an oscilloscope to check the PPM looks good?

tmrttmrt, 2015-08-23:
I have checked the pulses with a series TDS3000 oscilloscope. Statically they appear OK. When I quickly (in about a second) move a stick full range I only observe about 10 different pulse lengths in the persistence mode. The oscilloscope can acquire up to 450 waveforms/s so it is not due to oscilloscope, but due to low update frequency of the pulse data.

kilrah, 2015-08-23:
Can you look at the update rate of the whole pulse train? I.e is the problem that the signal is only sent out 10x per second, or that it's sent about 45x a second as it should be but that several successive pulse trains are identical?

tmrttmrt, 2015-08-23:
The pulse train update rate has period of 22.5 ms, as it should be.

tmrttmrt, 2015-08-23:
I enabled a pin toggle in evalMixes() function. It is called only every 113 ms! Why?

tmrttmrt, 2015-08-23:
Disabled lcdRefresh() and the period dropped to 22.6 ms. I am using ST7920 driver.

tmrttmrt, 2015-08-25:
I have fixed the ST7920 driver. I would need permission to create a branch to upload the modfications.

griebd, 2015-08-25:
Just make a fork, create a branch and make your changes. Then you create a pull request from your branch to OpenTX. @bsongis (most probably) will then check your pull request, commit it, and then give you permission on OpenTX directly, so you can create the branch here and delete your fork.

griebd, 2015-08-25:
and be sure to read https://github.com/opentx/opentx/wiki/Github-issue-workflow :wink:

tmrttmrt, 2015-08-25:
I have created a branch with a fix to this issue: https://github.com/tmrttmrt/opentx/tree/tmrttmrt/Fix%232750_Mega2560_ST7920_mixer_freq_too_slow. 
Basically writing to ST7920 is so slow that chokes up the main loop. Isplit  lcdRefresh() for this case in five stages to keep the loop running fast. Probably it would be better to attach the mixer calculations to some interrupt?  
I would prefer to wait some time to see if something shows up before a pull request.

tmrttmrt, 2015-09-01:
I have created a pull request. I do not understand git so there is one strange empty merge commit at the end. Please look at the next to the last for what has been changed.

projectkk2glider, 2015-09-03:
@tmrttmrt Your pull #2823 is against `next` branch. That is not right, you should work with `master` branch and create pull requests against it.

Now the easiest thing to do is:
* pull current master branch
* create a new branch for your changes (name should be different from your existing branch)
* now open log for your old branch and **cherry pick** the commits that you made.
* resolve any merging conflicts (if there are any)
* push commits to github and create a new pull request.

tmrttmrt, 2015-09-03:
I followed the instructions from https://github.com/opentx/opentx/wiki/Github-issue-workflow. There it is written: *He then creates an issue branch in the remote master repository by branching the branch called next.* ? 

I probably do not understand the git terminology or Github-issue-workflow needs an update?

kilrah, 2015-09-03:
It actually depends, shortly after a new major release like now we work mainly on master, then when things get stable we move on to next.

sloped-soarer, 2015-10-19:
@tmrttmrt
I'm using a LCD configured for SPI. I use a USART in MSPI mode, the refresh takes about 5.5ms, period 26ms. 113ms in your case seems an awful long time.
The LCD update should be a low priority task. I'd much rather have the mixes updated asap.
Something i should do is chart on a big piece of paper all the processes and their times. This code is new to Me ... still finding my way around it.

tmrttmrt, 2015-10-19:
ST7920 is awfully slow internally. One needs to wait about 50 us for each byte written. The pull request #2828 is just the simplest possible fix. It would definitely be better to put mixer calculations on a separate "thread" (timer5 on mega2560), but that would need a major redesign of how things run also for other boards.  On the other hand, most of the displays seem to be fast enough. Moreover, mixer calculations can become quite long with many mixers  (Can they become longer than 25 ms?) an have to be a kind of synchronized with pulses. A lot of headache for an 8 bit dying-out platform since ARMs are getting more and more accessible (Arduino Due for example).

JivaroFAD, 2015-10-20:
The testing of this pull request is ok (https://github.com/opentx/opentx/pull/2828#issuecomment-149517298).
Could we close this issue ?

bsongis, 2016-12-26:
This one seems fixed now

closed 2016-12-26 by bsongis