opentx/opentx #2754: Special Functions do not trigger on Switch Q (possibly others) on X9E

sky-flix, 2015-08-23:
When programming a special function to, for example, play a sound track from a switch, the switch can be selected by triggering it, however after programming is complete, the special function does not fire. The switch otherwise works fine for inputs.

SQ is an additional switch that I added to the X9E.

projectkk2glider, 2015-08-23:
Confirmed from sources. We have 56 possible physical switch positions on 9XE (18 x 3 pos + ONE + ON). Add to that all their negative states and we have more than 100 indexes.

The type `MASK_CFN_TYPE` which stores switches state is only 64 bit. Too small.

kilrah, 2015-08-31:
`MASK_CFN_TYPE` is used to hold *which functions have a switch set*, i.e. not related to physical switches. So 64 bits are OK (we have 64 SFs). Name might not be the best though.

I actually don't see what the problem can be right now. 
One issue is that the switch is stored in an int8_t and that is not enough anymore, but it currently only breaks if switch > FM6 or <!FM7 so doesn't explain why physical switches don't work.

The problem must be way earlier where switches are read from the hardware, which I may just have confirmed as if I install SQ on my board and toggle it it's SF that moves on the main view.

sky-flix, 2015-08-31:
I can't reproduce that here on a production X9E. My switch is plugged into the port silkscreened as SQ, and SQ toggles on the main LCD, and reads correctly for inputs, etc. I haven't tried using the SQ-down.wav and SQ-up.wav option in sounds/en/model_name/ yet to see if that works correctly. Will do that today.

kilrah, 2015-08-31:
Cool, thanks for the feedback. I have old proto boards but AFAIK layout hasn't changed.

MarenB, 2015-08-31:
I will continue to post here.

My related problem I described in #2816.
The switch works fine hardware-wise. The reading is correct in general and it is displayed on the main screen perfectly.
However, within the Global Functions it fails in certain settings. If possible, I will make a short video later.

MarenB, 2015-08-31:
https://youtu.be/4_ZJRQsEt64

should explain everything.

bsongis, 2015-09-01:
Reproduced here with SR

commit a985d021 (2015-09-01): Fixes #2754
commit 29e2ce7b (2015-09-01): Fixes #2754
closed 2015-09-01 by bsongis