opentx/opentx #2756: Logical switch with calculated sensor not working properly

lichtl, 2015-08-23:
Model: Taranis X9E
Firmware: opentx-taranisx9e-timer3-noheli-lua-mixersmon-ppmus-en-2.1.2N427.bin

I'm using a calculated Sensor to report the lowest cell (coming from my Cleanflight Naze). For reporting I want to use a logical switch using a<x which don't seem to work (L2 active all the time).

I've tried multiple settings and L2 only deactivate once I set x to 0 or into negative. Please let me know if you need any more data

[image]
[image]
[image]
[image]

Config file. Please rename to config.eepe
[image]

kilrah, 2015-08-24:
Did you create the "Cell" sensor on the radio or in companion? Was it done right now or does it date back some time? At some point there was a bug where a property wasn't set correctly.

lichtl, 2015-08-24:
I've created the "Cell" sensor at the radio and the testing was done at the time I've created this issue (The picture mention N425 but I've also tested with reflashed N427 )

lichtl, 2015-08-24:
Just another question. Is there a function to report the number of cells back? I've just rechecked on Companion but it don't seem to report back the number of total cells. (can split this also into a different issue/feature request)

AndreasAusD, 2015-08-26:
I have a similar problem : two voltage sensors, one FLVSS giving me a total voltage "FLVS" and a JLog 2.6 connected to a Kontronik Jive giving me a total Voltage "Vfas" 

Calculated sensor "diff" which adds "FLVS" and "-Vfas and a logical switch on "|a|>x" which never changes, no matter what value x has

Values differ by -0,07V and 0,03V when both connected and go up to 15V when only the FLVSS is connected to the battery

Tested on a 9xD with the release version of 2.1.1, no nightly build
all configuration done on the radio

joehallenbeck, 2015-08-27:
That's definitly the same problem I discovered. Try to relocate the telemetry sensor to a slot greater than or equal to 16. It should work properly then.

joehallenbeck, 2015-08-27:
[image]
This is my current working configuration.

kilrah, 2015-08-27:
Closing that one, #2787 has more detailed info.

closed 2015-08-27 by kilrah

AndreasAusD, 2015-08-28:
just tested it, diff on sensor 14 works, copied to sensor 25 does not work...
glad you found that bug