opentx/opentx #2761: Sticky Duration + And

pinkywafer, 2015-08-24:
I remember this being discussed before in #1646 back before the 2.0.10 release.
The sticky is now turned on by V1, and off by V2 and the AND only affects the whole LS switch, not the sticky, so for example
L1: Sticky  SHv  SHv  SA
after SH has been pressed once, SA will turn the LS on and off
but with duration, the sticky is reset also, so the above behaviour changes the AND (SA) will no longer trigger the LS as the sticky part has been reset.  I can't see the point of duration resetting the sticky, as in that case ie:
L1: Sticky SHv  SHv     duration 0.5
with current behaviour, the LS is turned on by the switch and off by the duration.  It cannot be turned off by the V2 switch ever, as duration keeps the switch on, so a sticky is not relevant, as the exact same behaviour can be found from:
L1: AND SHv SHv   duration 0.5
and also from:
L1: OR  SHv SHv  duration 0.5

Any thoughts please?

kilrah, 2015-08-24:
Not sure what the problem is. Is there anything this behavior prevents you from doing?

fraternl, 2015-08-25:
L1    a>x   Vfas    12.5
L2    a<x   Vfas     10.5
Sticky   L1  L2   duration 0.1

Yes...
I would expect this from the previous code...
The battery is 12.8V and L1 turns on the sticky, but because it has a duration of 0.1s this will last only 0.1s, but the sticky is still active and doesn't change so it should be off all the time
It will stick to true and L1 will not matter anymore. Only if L2 becomes true it will be possible to engage that sticky again.

In its current implementation it just flickers all this time until L2 becomes true
I would need another line with Edge to get what I want.

L1    a>x   Vfas    12.5
L2    a<x   Vfas     10.5
L3    Sticky   L1  L2
L4   Edge  [00:<<]   L1

kilrah, 2015-08-25:
What's the purpose of the duration there? Require the condition to be true for that amount of time before activating your sticky?

fraternl, 2015-08-25:
The duration acts just like the AND...   It is true only for 0.1s but the sticky stays on

kilrah, 2015-08-25:
That's normal, duration is "will be true for at least X"

closed 2015-08-25 by kilrah

fraternl, 2015-08-25:
Please keep this open....
I'm giving you a valid example

fraternl, 2015-08-25:
I know it would behave different than used with "AND", but that's not wrong is it?  
When I first tried to use duration I didn't expect it to flicker at all... I expected it to become true for that "duration" and never stop.
We could have that behaviour here?

kilrah, 2015-08-25:
Please give an example of what you want to do and why this doesn't work for you.
The sticky might not work the way you want, but the solution to get what you want might simply use other tools.

fraternl, 2015-08-25:
I want to create a reset of "Cnsp" when the battery's voltage is more than 12.5V, but after that reset is given, it shouldn't do that anymore.
The voltage might drop to 11V  for a moment and come back to 12.6
That's why I created the sticky
Only when the Voltage drops to a certain low, the Cnsp can get a reset again if it reaches 12.6V

I can then use the persistent option for Consumption
I think this would do it for a 3S batterypack

L1 a>x Vfas 12.5  delay 3s
L2 a<x Vfas 10.5   delay 3s
L3 Sticky L1 L2
L4 Edge [00:<<] L1

fraternl, 2015-08-26:
I just read https://github.com/opentx/opentx/issues/1646

There it was Bertrand's conclusion that the "ÄND" should not influence the flipflop.
He therefore chose for a different behaviour of "AND"
With the same reasoning the duration should not influence the flipflop itself either.

pinkywafer, 2015-08-26:
The problem in my mind as it is, is that the sticky ceases to be a sticky if a duration is used, as the duration resets the flip flop, so v2 is never evaluated, so you could use an and, an or as I showed above, I did miss the obvious "edge" though (I was tired!) which is a much simpler method of achieving the same result.

if I use:
L1: a>x  Alt  200
L2: a<x  Alt  100
L3:  a<x Vspd  0
L4: Sticky L1  L2  AND:L3
SF1: L4  PlaySound  Beep1  1s
SF2: L1  PlayValue Alt  Once
SF3: L2 PlayValue Alt  Once
If Alt more than 100 (having previously been over 200), I get a beep every second when descending to 100.  If I ascend within this window, the AND turns off the switch.  when I start to descend again, L3 becomes true and triggers the SF again.  SF2 + 3 read out at just over 200 and just under 100
I like this, but decided that I only wanted this alert for 2 seconds of the descent, as I may be making a slow descent, and not want the constant annoyance, so changed L4 to:
L4: Sticky  L1  L2  AND:L3  Duration 2
Now I have lost the functionality.  After the duration, the switch cannot be reset by the AND, and I'd have to climb to 200 again to reset.

This doesn't prevent me from using it, as I can remove the duration, then use an additional LS:
L5: Edge L4 [0.0:<<] Duration 2
but I just can't see why:  I can't see the point of a sticky which can never be sticky: duration prevents a sticky from ever being sticky.  For the same reason as the AND was taken out of the flip-flop reset, I would vote for the duration to be taken out and applied only to the output of the whole switch, the current behaviour does not need a sticky, and could be done with various other types as V2 is never evaluated to turn off the flip flop when duration is used.