opentx/opentx #2762: 2.1.x Simulator does not populate telemetry data causing constant alarms

jtaylor2, 2015-08-24:
Tested Companion Taranis simulator and standalone Taranis simulator with self compiled 2.1.0, 2.1.1, and 2.1.2 compiled on 64 Bit OpenSUSE 13.2, 32 Bit OpenSUSE 13.2 and Windows 7 POSIX compiled with Cygwin.  Problem does not exist with 2.0.17 on any of these platforms.

Problem:  (Only tested with D series receivers) The simulator does not populate the radio detected and populated telemetry sensors with any simulated data causing constant alarms if alarms are programmed using LS, SF's.  Using the model from the example below, as soon as the simulator is started it says
[image]
 A2 Critical every 10 seconds.

If the F4 telemetry simulator screen is brought up and data is entered it does not go into the existing sensors, but new sensors are created and the data is populated into them and the alarms continue.
```
Example:  Model with 4 auto detected sensors from d6fr receiver
RSSI ID    F101    25
A1         F102    25
A2         F103    25
SWR        F105    25

None of these are populated with data and the following LS/SF alarms constantly:
L1    A<x    A2    9.69v
SF11    L1    Play Track    a2_red    10s

Bringing up F4 Telemetry simulator and populating with data does not populate the existing
 sensors, but creates 4 new sensors below them and populates them with data and the alarm
 continues (See screen shot).
.
RSSI ID    F101    2
A1         F102    2
A2         F103    2
SWR        F105    2
```
Jim

projectkk2glider, 2015-08-30:
The *Telemetry simulator* currently only simulates the S-port type sensors. So it is no wonder to see what you are seeing.

We could add an option in the telemetry simulator that will set the type of telemetry sent.

jtaylor2, 2015-08-30:
@projectkk2glider  Yes, but the telemetry simulator works at 2.0.17 with this configuration and even without it the old way of setting A1 and A2 alarms didn't cause them to alarm constantly in the Companion radio simulator (with no way to stop the alarms) like it does now.

In this regard I think 2.1.x is a step back in usability form 2.0.17 

I think there are two things that need addressed.
1. The telemetry simulator needs a way to set which sensor instance ID it is simulating (either for each sensor.....a lot of work) or for all of then.
2.  The Companion Taranis simulator needs to populate the sensors with reasonable data at startup so that it doesn't alarm constantly.

I'm currently out of town and won't have any time for a few weeks, but if nobody does anything with it I'll see if I can figure out the code enough to propose a solution when I get some time.

projectkk2glider, 2015-08-31:
It worked in 2.0.x because both sensor types filled the same telemetry values. Now in 2.1.x things are much different. Not sure what the best solution is right now.

kilrah, 2015-08-31:
The telemetry simu could load the sensors from the model data...

jtaylor2, 2015-09-03:
I have created branch jtaylor2/issue_-2762 (sorry for the extra - in the name) that has changes to allow changing the  sensor instance that the telemetry simulator uses to send packets.  This allows sending packets to the instance 25 that the D series sensors come in as.

I have also added a change that will populate existing sensors in the Companion and standalone simulators with RSSI, SWR, A1, A2, and RXBat default values.  This will prevent constant alarms if these values have alarms set using LS/SF's.

These changes seem to correct the problems I reported in this issue.

Please take a look at branch jtaylor2/issue_-2762 and if the changes look OK and you think it might be useful I can generate a pull request.

commit 2bf3f803 (2015-09-06): Merge pull request #2834 from opentx/jtaylor2/issue_-2762

Telemetry simulator improvements (fix #2762)
closed 2015-09-06 by kilrah