opentx/opentx #2774: Output gitter - visable on mixer screen and RX output

ChrisParish, 2015-08-25:
I am experiencing quite a lot of gitter on the outputs from the RX. On closer inspection the gitter is visible on the channels and mixer output screen as a bluring of the last digit. This is the case even on channels which are sourced from a switch if that switch is then applied through a curve.

I use various high channels as mix channels and when they are applied to the RX output channels via curves I then get the gitter. I would speculate that the problem is related to the output from the curves.

Running 2.1 on X9E and 2.1.1 on X9D and both are the same. Down-grading the X9D to 2.0.X fixes the problem.

kilrah, 2015-08-25:
Strange... Please upload your settings file to a file sharing site and post the link here!

ChrisParish, 2015-08-25:
Here you go:
https://www.dropbox.com/s/9gnuzfe9pk7bqn7/gitter.eepe?dl=0

kilrah, 2015-08-25:
Thanks, can you also tell us what particular channel to look at?

ChrisParish, 2015-08-25:
The jitter is usually evedent on channels 1 to 6 and 32. Channel 1 is has a switch as the input and uses a curve for shaping the output so I would expect to see no jitter on this as the input is either -100 or 100

ChrisParish, 2015-08-25:
I have created a quick video to illustrate the problem:

https://youtu.be/YjDNc9IfK5w

kilrah, 2015-08-25:
There was definitely something wrong with CH1 and I think I got it. The others are normal analog jitter from what I can see.

ChrisParish, 2015-08-25:
What is strange is that the jitter is much worse with the 2.1+ firmware. Going back to the 2.0.X on the X9D fixes the jitter. Obvously I cannot do that with the X9E.
The other channels which are giving me greif are also not sourced directly from a stick, but via another channel. 
It appears that the greater pecentage of the channels input that comes from a mix channel or non-stick source then the worse it is.

kilrah, 2015-08-25:
Can you try this?
http://www.andrebernet.ch/divers/opentx/opentx_issue2774.zip

ChrisParish, 2015-08-25:
That just drove my TX a)french and b) mad.
Did you compile for X9E?

kilrah, 2015-08-26:
Ah sorry, that was for an X9D+, didn't notice you only had a non-Plus.
Wait for the review of #2775 and try the nightlies if merged then.

jjmouris, 2015-08-26:
Hi all, i am the one with a X9D+ but the above zip file now seems to no longer be there. For the moment i have gone back to 2.0.17 as this issue is very noticable on my airframes and i don't want to wear out the brushes faster on the very expensive servo motors (Hitec 7115). Have to say, the rest of 2.1.1 looks very nice indeed. 

For reference, the issue was first reported here by me.
http://www.rcgroups.com/forums/showpost.php?p=32514006&postcount=817

ChrisParish, 2015-08-26:
I have the file if you want to try it. 

Ps. Fight over who first reported it?? Really?

;)

C

kilrah, 2015-08-26:
File is back in place

jjmouris, 2015-08-26:
No fight, was just to link in the thread. :)

ChrisParish, 2015-08-26:
If I deselect the trim option from all mixer which are sourced from a non-stick source then the problem goes away. Looks like you are onto something.

fraternl, 2015-08-27:
So that's why I never noticed that.
Not really a user error, but these shouldn't be enabled...
I hope it will now get a fix soon

kilrah, 2015-08-27:
Should be fixed in next nightly.

closed 2015-08-27 by kilrah

jjmouris, 2015-09-07:
So what is the status of this problem? I see the thread is closed? But this issue is still alive for me untill it is incorperated in V2.1.2 or what?

kilrah, 2015-09-07:
The fix has been coded and merged into the master branch, which means next release (2.1.2) will include it.

jjmouris, 2015-09-07:
Any idea when 2.1.2 will be released?

kilrah, 2015-09-07:
Very shortly.

commit ed2ae75c (2015-08-27): Fix #2774 (out of bounds access to trim array when the source of an input is not a stick)