opentx/opentx #2776: Different "duration behaviour" when used together with Sticky

fraternl, 2015-08-26:
This was discussed in http://openrcforums.com/forum/viewtopic.php?f=45&t=6063#p85696, in https://github.com/opentx/opentx/issues/1646 and more recently in https://github.com/opentx/opentx/issues/2761

Sticky is a flipflop
Its power in usability derives from the fact that the input for this flipflop is V1 to turn on and V2 to turn off. The optional AND does not turn off the flipflop

@bsongis even changed the behaviour of "AND" for this.

That same reasoning could be used to either turn off "duration" or change it to what I would expect from duration before I started to use it in OpenTX.

The duration could be changed into the period that the LS is true while the flipflop is true.
So a duration of 1 second would mean that the LS is true for 1 second after the flipflop is turned on. The LS would be turned off after that, but the flipflop would remain active until V2 turns TRUE.

I expected the duration to do this, so I wrote this as a means to reset Cnsp when a full battery is attached to my receiver.

L1 a>x Vfas 12.5 delay 3s
L2 a<x Vfas 10.5 delay 3s
L4 Sticky L1 L2 duration 0.1s

It doesn't do what I want because the duration doesn't do what I would like it to do.
duration is normally the minimum time the LS is active. 
Because sticky is used to "stick" a value, turning a momentary input into a latching switch,  this duration in that form feels off.
IMHO it just ruins the flipflop and in its current form I can't see any usability for duration (please correct me if I'm wrong).
If the duration would just be a property of the LS as a whole and not of the flipflop, I could use the previous example.

This example is working and was what I was aiming for:

L1 a>x Vfas 12.5 delay 3s
L2 a<x Vfas 10.5 delay 3s
L4 Sticky L1 L2 duration 0.1s

In https://github.com/opentx/opentx/issues/2761 this issue was closed without being discussed properly. An example was asked and I gave one.
Still it got closed....

Can someone give me an example how duration could be put to use in its current form??
As @pinkywafer pointed out the whole idea of sticky is just ruined by it.

However, the "duration" does..

fraternl, 2015-08-26:
To add to this and hopefully clarify...
I propose that duration in Sticky is used to prolong/limit the time the LS is.

If the flipflop is true for 10s  and duration is set to 1s, the LS should become active for 1 second.
if the flipflop is true for 1s and duration is set to 10s, the LS should become active for 10 seconds

kilrah, 2015-08-26:
You could have continued discussion in the other issue, it would have been reopened once given an explanation we can understand.

Now I got it. There probably isn't a purpose in Duration right now indeed, but how it works is just a consequence of the way it's implemented globally for all LSs (i.e. a "special case" would be needed in how Duration is handled for the sticky, which was never done).

fraternl, 2015-08-26:
Hi Kilrah... 
I don't have that much experience with github and its etiquette...
I assumed the discussion became somewhat closed if the issue got closed.

Reading your post I think you're inclined to keep it the way it is although you agree it is probably not being used at all as the behaviour is almost unpredictable.
Wouldn't a remark on the explanation of Sticky, just like it's done for the "AND", be enough?

tomkfly, 2015-08-26:
The way I see it:- If the trigger is no longer true duration will unlatch the sticky after the specified period, if the trigger condition is met again, it will repeat until the unlatching switch condition has not been met. I believe that is the way it should be

kilrah, 2015-08-26:
The problem is that the LS is kept ON for Duration as soon as it's turned ON (regardless of what happens with the 2 conditions), AND that it's always pulsed OFF at the end of Duration also regardless of the 2 conditions. It's indeed useless as it is, it probably simply wasn't considered to use it on a Sticky. 
Now whether Duration is simply disabled or behavior is changed...

>Hi Kilrah...
I don't have that much experience with github and its etiquette...
I assumed the discussion became somewhat closed if the issue got closed.

You should know from past experience with your many issues ;)
Now we have discussions in 2 places and it's a pain...

Spainman, 2015-08-26:
I vote to keep it the way it is.
The flipflop of the sticky switch should not have an internal, invisible state that is undiscoverable by the available means. The AND condition acts in line with it's general purpose (additional condition) and is visible because it's external to the switch. Duration is internal, invisible and part of the sticky switch as an object from the users external point of . Assuming that duration would not reset the flipflop there would be the situation where all visible state information suggest that V1 should trigger the switch upon the next actuation but it wouldn't. That's no good.

Regarding the question if duration is of any use in combination with sticky: Yes, it is.
I consider it as a flipflop with an additional time based unlatch functionality and use it at several occasions in complex esc arming sequences and in remote engine start.

fraternl, 2015-08-26:
@Spainman on openrcforums you wrote:

"A fliplop whose 'set-ability' is dependent on the reset-reason and not on it's current value goes beyond my imagination of possible use cases."

and...

"Considering that the duration period can't be interrupted, I agree. Maybe duration shouldn't be available with sticky anyway like the n/a delay on edge."

But of course, you are allowed to change your mind (or as you say, find a creative way to make use of it)

Spainman, 2015-08-26:
I did not change my mind regarding the first quote, maybe misinterpretation:
The first quote boils down to what I wrote here and the behaviour of duration was changed to what is now after that argument - If the sticky switch is in non-set state from the users point of view, triggering V1 should turn it on, no matter if duration or V2 did the reset. Actually I went beyond that and asked that AND should also reset the sticky switch but changed my mind in course of the discussion after understanding the role of that parameter.

Regarding removing the duration parameter from sticky - yes, I found uses. Actually I even pointed that out in the very same thread in discussion with kilrah: 
"- Having 'Duration' alter the state of the flipflop would be useful for example for the timed pulse explained above. I think it would be reasonable to have a timed self deactivation without impeding the next activation. It would add usefulness and open possibilities without having to work around the strange reset-twice effect, but I know that we two disagree on this..."

fraternl, 2015-08-26:
"but I know that we two disagree on this..."
I don't necessarily disagree....
I was under the impression that the current behaviour doesn't make real-world things possible. You say you have and I can't debunk that, 

I already had to get use of using the duration in the normal switches.
Of course we can't change that.

fraternl, 2015-09-01:
Due to myself we now have 2 tickets discussing this. In the old (closed) ticket (https://github.com/opentx/opentx/issues/2761) there are those that are in favour of changing it.

If Spainman really found a practical use for that duration it's better to keep it that way because it works as the other switches do.

Now that "delay" is turned off for sticky. Could thát be used for a maximum duration?

@pinkywafer :  please give your input in this ticket

Spainman, 2015-09-01:
What do you mean with "delay is turned off in sticky"?

fraternl, 2015-09-01:
Scratch that comment of mine.
It's in edge that delay is turned off....

I don't have my radio here and just didn't check...

Did I give you a scare, BTW?   ;-)

Spainman, 2015-09-01:
Obviously you did! ;-D
But now that you mention Edge and the missing Delay: I don't see a particular reason why Delay is not available. Basically a time offset actuation seems to me like a valid, useful option independent of the kind of switch.

kilrah, 2015-09-01:
Yes but it can't be done easily as Edge causes a single pulse that with a delay would have to happen some time after the triggering condition is already false again... which implies memorizing states which isn't required for any other LS type.

closed 2015-09-01 by kilrah