opentx/opentx #2777: Radio-wide volume control

pindeslandes, 2015-08-26:
Taranis.

I know this has been discussed in other issues but I couldn’t find what would be thought about a proposal like this one:

In the radio general settings, instead of just a Volume level, would it be possible to have:
- Volume source (give choice among physical inputs: pots, switches…),
- Volume range (min and max levels),

so that the volume level would be controled by the volume source?

No Volume source selected would only allow max volume setting (as it is now, and it would be the default configuration).

kilrah, 2015-08-26:
Volume global function?

pindeslandes, 2015-08-26:
I don’t think so because global functions start after tada.wav is played, so at least the beginning of tada.wav is played at default volume, which is annoying (and sounds funny too when the last part of tada.wav is played at low volume).

kilrah, 2015-08-26:
That is a consequence of the timing with which different things are handled.
i.e another way of choosing volume would behave the same, respectively it would be possible to delay tada.wav so that the current global function works.

pindeslandes, 2015-08-26:
I am not sure of that either. In such a case, I suppose you would hear any sound global function before tada.wav?
Anyway, the Volume global function would still miss the min/max level settings. The only way I am aware of is to define the same input or mixer (ranging from min to max desired levels) in each model and use it in the Volume global function, which makes it not so global.

fraternl, 2015-08-26:
I'm all for a delay of Tada.wav to only kick in after volume is set by a special function (global/local)

pindeslandes, 2015-08-27:
Well, I don’t know how much it would be difficult to create a volume control in the radio general settings (which seems the most natural solution to me) but, IMO, it would be simpler for the user and it wouldn’t have side effects as a global function may.

projectkk2glider, 2015-08-27:
In that case we would need following settings (in general radio settings):
* adjustment range of custom volume - ie when source is zero volume is test to the bottom range value, when source is at its max the volume is set to the top range value. 
* the source of volume adjust (list containing sliders and pots)

The per model defined Volume special function (or global function) would still be able to override this setting if defined and active.

pindeslandes, 2015-08-27:
Yes, exactly what I had in mind (plus: the source list may include physical switches).

closed 2019-11-15 by stale[bot]