opentx/opentx #2780: Lua getValue() not working for special telemetry values

projectkk2glider, 2015-08-26:
The #2773 reminded me that we are missing a way for Lua to get the values of all *special* telemetry sensors. That pull fixes the case for GPS coordinates. But other sensors types like cells and datetime also need this special handling. 

Also related to the request to have all individual cell voltages avalilable in Lua #1799

dsbeach, 2015-08-26:
I might be in a position to take this on.  Is there a list of 'special'
sensors, or should I just find all the enumeration values for
TelemetrySensor.unit?

projectkk2glider, 2015-08-30:
>should I just find all the enumeration values for TelemetrySensor.unit?

Yes that is the right path. I would also like to change the flow of `luaGetValueAndPush()`: there is no need to call `getValue(src)` for these special telemetry values. It might as well be done along with this issue.

dsbeach, 2015-09-10:
Should be closed by #2826

closed 2015-09-10 by kilrah