opentx/opentx #2782: V2.1 telemetry: easy setting of sensors alarms

pindeslandes, 2015-08-27:
The new telemetry system looks awesome but it seems the former Rx voltage alarms have gone. Now you must recreate them by spending logical switches.
Besides, with all these sophisticated sensors we can build now, I wouldn’t be surprised if the needs (or temptation…) to set much more alarms than ever were rising, along with a massive use of logical switches.

From where the idea to add a feature like this, as well to save logical switches as to make sensors alarms settings easier:
- Add an ‘Alarm threshold’ field to each sensor line.
- Automatically play *SensorName-Low.wav* (or *SensorName-High.wav*) when the sensor goes under (or over) the alarm threshold.

projectkk2glider, 2015-08-27:
Interesting idea, but it might require an eeprom change, in which case it will have to wait for 2.2.

pinkywafer, 2015-08-29:
This sounds like a great idea. As an eeprom change would be required (to store the threshold) could I suggest the addition of :
Low/high threshold field
A boolean to select low or high
And also a critical threshold field (in the same direction) 
A sound file field to select which sound file to use as the sensor name in the warning announcement. (I'm guessing this may be needed due to sensors being re-nameable/addressable)
That would allow automatic warnings for [sensor name] low' (or high) and '[sensor name] critical' without any logical switches or special functions being required

kilrah, 2015-08-29:
That would require MASSIVE eeprom usage, so no.
The original request just maybe, and you handle the low/high with the name of the file.

pinkywafer, 2015-08-29:
Ok, point taken

kilrah, 2015-08-29:
By the way I don't even like the original request, because it would add a setting in the UI that does nothing on its own, requires a computer and non-straightforward configuration to make use of. Would be highly confusing to users.
The current similar thing done for model name, switches etc is OK because it is completely independent from the UI so no possible misleading.

pinkywafer, 2015-08-29:
I can see what you mean. Trying to think of any alternative method of doing these alerts, without using lots of logical switches, and without having an extra ui setting which does nothing by  the only thing I could think of was to have a seperate 'sensor alarms' page with x alarms available which could be assigned with options: sensor, threshold, above/below, warning file. But I'm expecting that this would also cause massive increase of eeprom?

kilrah, 2015-08-29:
Yep, and for one limited use only that would not even cater to many. Personally my alarms are nearly never just "value < threshold", they're combined with other conditions. For example on my models with a current sensor (most) I have a "current >2A" LS that I use to gate my capacity and other alarms, so that they automatically disable themselves when I land and turn the motor off.

To me what makes most sense is to increase the number of LS, at least the extra memory will also serve other purposes.

pinkywafer, 2015-08-29:
Yes, you're right. More logical switches would do the job, whilst also being available for other uses

bsongis, 2015-08-29:
The 2.2.x will bring 64 logical switches per model

3djc, 2017-02-19:
I think by the time you've added all the things you need to get it really right and usefull, you have redefined a logical switch. I personally think this one should be closed rather than tagged 2.2.1

closed 2017-08-15 by 3djc