opentx/opentx #2783: Add MIN and MAX to mixer’s multiplex functions

pindeslandes, 2015-08-27:
Of course, it’s easy to do with ‘a<b’ LS but it would make the mixer faster to set, easier to read and to debug, and it would save logical switches.

projectkk2glider, 2015-08-27:
I don't understand. Please provide an example.

kilrah, 2015-08-27:
Min/max of what? 

When a mixer line is evaluated there is no way to know the final channel value, there may still be other lines after it.
Also, line 3 cannot know what value line 1 had, it can only know what the result of line 1 and line 2 is (intermediate calculations not kept).

Simply not possible with the processing pipeline we have.

pindeslandes, 2015-08-27:
I meant: min/max of the current line and the result of the previous one (just like Multiply, for instance).

kilrah, 2015-08-27:
OK, much clearer. Why not, let's discuss, need to check if we have enough free space in the mixer structure which I doubt.

closed 2019-11-15 by stale[bot]