opentx/opentx #2787: Telemetry sensors > #16 not usable as logical switch

joehallenbeck, 2015-08-27:
Logical switches do not accept telemetry sensors greater than #16.

I'm developing my own minimOSD firmware, sending back a lot of telemetry data. Currently I'm using 24 of the available 32 telemetry slots. I discovered that only the first 16 telemetry slots are usable as logical switch.

kilrah, 2015-08-27:
Is that only a problem when programming and transferring to the radio from companion? Does it work when set up on the radio only?

joehallenbeck, 2015-08-27:
No. It makes no difference wether setting it up on Companion or the Taranis itself. I figured out the problem by trial and error. Logical switches using telemetry sensors over slot 16 doesn't work properly.

joehallenbeck, 2015-08-27:
I'm not a native english speaker, sorry for that. But I can provide screenshots or a video describing the problem. It's not grave, but I thought it might be something you should know.

kilrah, 2015-08-27:
OK, most likely explains #2756

joehallenbeck, 2015-08-27:
I forgot: I'm using OpenTX 2.1 on a FrSky Taranis Plus.

kilrah, 2015-08-27:
Bleh. LS evaluation fails when cs->v1 > 255, I found a place that would definitely cause it, but it still fails after fixing. Can't seem to find another similar mistake in the data path.

kilrah, 2015-08-27:
Addendum, as described companion also breaks on import/export of values >255.

closed 2015-08-27 by kilrah

kilrah, 2015-08-27:
2nd one was hiding, thanks @bsongis !

joehallenbeck, 2015-08-28:
Thanks for fixing this so fast.

commit b22df02d (2015-08-27): Fix #2787
commit 34f7c3a4 (2015-08-27): Fix #2787