opentx/opentx #2788: RF module state from the upload disabled

TaranisHarry001, 2015-08-27:
Hello,
have played on my Taranis 2.1.1 and also the accompanying Companion 2.1.1 installed on the PC. I noticed when uploading from the transmitter to the PC that the RF module is always deaktiveirt. I change this setting under Companion and then write them into the Taranis and I read this back, then everything is ok.

Harry

bsongis, 2015-09-01:
I think you installed a beta OpenTX 2.1.0 and your model were converted there, right?

TaranisHarry001, 2015-09-01:
Hello,

first, i have installed the beta opentx 2.1.0 and then i have flashed the version opentx 2.1.1 and compansion 2.1.1 on my pc.

bsongis, 2015-09-01:
Ok then it's normal, the bug was an import bug in the first opentx 2.1.0 beta versions. Not a huge issue, but you have to fix your module settings on Companion as you do and everything will go fine after.

TaranisHarry001, 2015-09-01:
perfect. thanks. good night.

kilrah, 2015-09-01:
No, it's still a problem right now!

bsongis, 2015-09-02:
The eeprom was also badly converted in a previous 2.1.0 beta version. I could fix it, but Companion would be unusable for those who have the internal module hack (they have PPM sent on the internal module instead of PXX)

closed 2015-09-02 by bsongis