opentx/opentx #2790: 9XE power on/off loop possibility

projectkk2glider, 2015-08-27:
The recent commit 77d3eae4d608f4a65fee786d4f31771daec36452 from the pull #2772 which was done to help battle the RTC reset issue #2626 can/is causing problems if the additional capacitor on VCC line is too big. After the power off, the radio will go into a cycle of power on - power off (couple of times a second).

Let me explain what is happening:
* The radio is on
* The user presses the power button and holds it enough time to trigger the power down.
* the `boardOff()` function is called
* he MCU_PWR_ON output is driven low.
* while the power button is still pressed the watchdog is periodically reset, when the power button is finally released the code continues.
* in order to fight the RTC reset issue the radio now enters the low power mode (a recommendation/solution by Mike). In this mode the watchdog is not poked (reset)!
* now comes the gist of the problem:
  * the radio by now is without battery power, the FET connecting the battery to the switching power supply is OFF.
  * the VCC is falling, the rate of fall depends on the radio consumption and the value of additional capacitor.
  * if the **rate of VCC fall** too smal, then the watchdog resets the radio **before** the VCC falls below the brownout reset value.
  * this reset then restarts the radio firmware and the first thing after the reset is the turning on of the MCU_PWR_ON output which then connects the battery to the power supply. The radio power is restored.
  * then later radio finds out that power switch is not pressed and starts to turn of by calling `boardOff()`. 
  * the cycle repeats.
 
This issue is not happening on stock 9XE development board, but is shows itself if I solder 1000uF capacitor to the VCC line. I haven't tried other values. FrSky plans to use 470uF. The tolerance for electrolytic caps is huge (like 20%). So some radio could be affected and some not.

There are two possible solutions:
* to (again) disable the part of the code that puts the CPU in low power mode.
* to devise a system to poke the watchdog even in low power mode.

kilrah, 2015-08-27:
>  the radio now enters the low power mode (a recommendation/solution by Mike).

Yeah that was actually already applied weeks ago and quickly removed due to that problem. Didn't see the fix he posted the other day was the same.

We can simply set the watchdog to a ridiculously long timeout when we get to that point, like is done in flashSportDevice().

projectkk2glider, 2015-08-27:
>We can simply set the watchdog to a ridiculously long timeout when we get to that point, like is done in flashSportDevice().

That was my though also, but a was not sure if the watchdog reset period can be modified once it is configured. Apparently it can be. I will look into it. If we can set ti to one second or more we are golden.

bsongis, 2015-08-27:
In per10ms there is already a trick for the WDT. Perhaps it will help here
if needed.

kilrah, 2015-08-27:
Mmh that's what I've just seen, and no it's unfortunately the opposite of what we want as it's a software trick around it, but software doesn't run at this point due to sleep mode.

projectkk2glider, 2015-08-27:
Exacly the watchdogSetTimeout() is not useful here. I am looking at reconfiguring the watchdog reload value before entering low power mode now.

projectkk2glider, 2015-08-27:
Bummer, it is not the low power code that causes problems! Even with that part disabled I get power on/off oscillations (with 1000uF cap). I guess I need to hook up to oscilloscope.

For the watchdog, the reload value can be changed, it can even be disabled in this case (if needed) by stopping LSI clock! But looks like the problem is elsewhere at the moment.

projectkk2glider, 2015-08-27:
Some measurements, the yellow line is VCC, the blue line is NRST signal.

First a normal power down without additional capacitor:

[image]

Now with 1000uF cap (voltage and time settings the same):
[image]

Measuring time, voltage from the moment the power is turned off to the reset signal activating (9.6ms):
[image]

Measuring the time and voltage to the first restart (NRST goes high again)
[image]

Strange thing is the radio resets, then NRST rises, then it resets again for a very short time then it powes up and the MCU_PWR_PIN gets driven high. I have no explanation for this behavior for now. But the watchdog is not responsible, the whole event takes only 40ms. I have also programed BOR to the VBOR3 already, and the behavior is the same.

MikeBland, 2015-08-27:
I'll have a look myself when I get some time. I'll need to add a big cap though.
We had a problem with the processor getting powered from the RTC battery, due to using the different chip requiring different connections. It might be worth repeating the test with the RTC battery removed.

What is odd, is that when NRST first goes high, it actually reaches a higher voltage than the VCC voltage, which is what makes me think it is charging up towards the RTC battery voltage.

Mike.

MikeBland, 2015-08-27:
From the datasheet, section 2.2.15:

"Internal reset OFF
This feature is available only on packages featuring the PDR_ON pin. The internal power-on
reset (POR) / power-down reset (PDR) circuitry is disabled with the PDR_ON pin.
An external power supply supervisor should monitor VDD and should maintain the device in
reset mode as long as VDD is below a specified threshold. PDR_ON should be connected to
this external power supply supervisor. Refer to Figure 7: Power supply supervisor
interconnection with internal reset OFF.
Figure 7. Power supply supervisor interconnection with internal reset OFF
(Image here)
1. PDR = 1.7 V for reduce temperature range; PDR = 1.8 V for all temperature range.
The VDD specified threshold, below which the device must be maintained under reset, is
1.8 V (see Figure 7). This supply voltage can drop to 1.7 V when the device operates in the
0 to 70 °C temperature range."

So we may need hardware assist to hold NRST low if a large capacitor is fitted.

MikeBland, 2015-08-27:
Possible software solution:
Make use of one of the (RTC) backup registers. When powering off, write a (non-zero) value to the register. On restart, in the bootloader, if the register contains this non-zero value, do NOT turn the soft power back on, clear the register, then go to sleep again.
Due to there being more than one exit from NRST being low, this may need to be more complicated.

There may be another method of checking the reset reason, together with checking for the power button being active, or a combination of all this.

The main thing may be to definiitely start on a watchdog reset, or the power button pressed.

projectkk2glider, 2015-08-28:
>It might be worth repeating the test with the RTC battery removed.

All my tests are without the RTC battery.

>What is odd, is that when NRST first goes high, it actually reaches a higher voltage than the VCC 

The voltage scale between yellow and blue line is not the same. I meant that the voltage and time setting is the same for all screen-shots. So the VCC is always higher that NRST voltage. Sorry for the confusion.

>The main thing may be to definiitely start on a watchdog reset, or the power button pressed.

That is a good idea, I will try that next.

MikeBland, 2015-08-28:
I can't reproduce the problem on my board. I've added a 1000uF cap (I already had a 100uF cap fitted so I have 1100uF).
Testing with a RTC battery fitted.
On power off, I see a 0.5mS low pulse on NRST, but both the NRST signal and the power voltage just decay away over something like 300mS.
When I power off, I get the 'lock' display, then the 'ZZ' display, then when I release the power button everything just powers off.
I'll try later without a RTC battery.

MikeBland, 2015-08-28:
Tested without RTC battery and I have the same result, all works OK.

If the processor is in standby (deepsleep) mode, then on a NRST restart, we have:
"After waking up from Standby mode, program execution restarts in the same way as after a
Reset (boot pins sampling, vector reset is fetched, etc.). The SBF status flag in the PWR
power control/status register (PWR_CSR) for STM32F405xx/07xx and STM32F415xx/17xx
indicates that the MCU was in Standby mode."

The SBF flag may be another way to detect we should be powering down and so not turn the soft power back on.

Please confirm you have pin 143 lifted and tied to VDD (pin 144).

projectkk2glider, 2015-08-28:
>Please confirm you have pin 143 lifted and tied to VDD (pin 144).

Yes I have.

Mike I have the issue solved. The MCU_PWR_PIN gets turned on in _bootStart() only if:
* the radio was reset by a watchdog or software reset
* or if the power button is pressed

That completely cures the power on/of loop. 

Now I am seeing another related issue. After the watchdog reset the top lcd gets corrupted. All segments on second controller (Ht1621Data2[]) are OFF. Still researching this one. This happens on latest master also.

projectkk2glider, 2015-08-28:
> After the watchdog reset the top LCD gets corrupted.

I have this solved also. I will make a pull with my fixes soon.

projectkk2glider, 2015-08-28:
The results from #2801:

Power down with 1000uF cap:
[image]

Power down with no cap:
[image]

Wathcdog reset (uses different time scale):
[image]

Looks good to me.

I will do some more tests. @MikeBland Mike can you test it too?

MikeBland, 2015-08-28:
Is there a built binary I can test?
With the cap fitted, I have a completely different response (maybe BDO setting).

As soon as the supply drops to 3.0V, I get a NRST, that lasts only 0.5mS, then the power voltage just continues to drop with the NRST signal rising back up to supply voltage then dropping with it.

MikeBland, 2015-08-28:
My option byte(s) at 0x1FFFC000 is AAFF, so no flash protection, no BOR, and reset generated on entering standby mode. If you have changed your BOR setting, have you also changed the other bits?

closed 2015-08-28 by projectkk2glider

MikeBland, 2015-08-28:
What are the option bytes you are testing with? I believe the default, and what FrSky are likely to ship with, is AAFF. I would suggest you make sure your option bytes are the same, and check the resulting response at power down. I understood you had set your option bytes to enable the BOR at BOR3 level.

I'm not convinced we need the BOR enabled. The core works down to 1.8V (or less), and the POR circuit operates at this level. All I see is the BOR will give a reset when the processor is quite capable of still running.

projectkk2glider, 2015-08-28:
I've programmed BOR back to default. Or did i? I only issued a flash command to program the old value back. Maybe it should be erased first?  I will check tomorrow.

MikeBland, 2015-08-28:
My understanding of what I see at power off is I enter standby, and the bit in the option byte causes a NRST to be generated. This does NOT reset the processor, which remains in standby mode waiting for an event (_WFE).
Since I don't have the BOR set either, the power just decays.
From your 'scope traces, I reckon you had the BOR enabled, and the standby reset bit disabled.

projectkk2glider, 2015-08-29:
I have now checked my option bytes by reading FLASH->OPTCR register. They are (lower 16 bits) 0xAAED which corresponds to the default value of option byte(s) at 0x1FFFC000 (AAFF). 

I also saw that I probably never changed my BOR level in testing because I forgot one crucial step in programming them. So all my tests were done with default option bytes.

I will prepare a build for you to test, actually two builds one before the fix and one with the fix.

projectkk2glider, 2015-08-29:
@MikeBland two testing builds for you:

* first one http://downloads-21.open-tx.org/test/opentx-9xe-build1.bin from the master 88aea7f78017c11a949729456a8449e38ab97f53 (before the fix was merged) and has a potential to do a power on/off loop (it certainly does it on my radio with 1000uF cap).

* second one http://downloads-21.open-tx.org/test/opentx-9xe-build2.bin from the commit a28f6e2a63706bdc0a8a661bdc6a2138c38c6ae2 which includes the fix and also displays option bytes.

Both are build with a command:
```
make PCB=TARANIS PCBREV=REV9E HELI=YES GVARS=YES LUA=YES WATCHDOG_TEST=YES OPENTX_VERSION_SUFFIX="wd-loop1"
```

To display option bytes (available only in second build) go to DEBUG menu (main menu, long press ENT, Statistics, scroll). Long press ENT key in DEBUG menu to refresh it (it will be 0 in the beginning, ti is only read with the long ENT press).

You can also do a watchdog reset test (in both builds) go to DEBUG menu. long press MENU then ENT.

Looking forward to your tests.

MikeBland, 2015-08-29:
Having flashed both of these, I'm now surprised as the decay time of the power rail is now very much longer, and stays that way even if I put ersky9x back on. On earlier tests, the power decayed over around 300mS, now it is taking several seconds.
All tests done with the RTC battery in place.

Both files work perfectly OK as far as power off is concerned, I haven't checked the watchdog reset yet.
This is what happens for me at power off (your second build), on both your builds and ersky9x of mine.

[image]

I haven't changed my option bytes so they should still be at AAFF.
This is on the second prototype board we were sent (the latest board I have). I have no case or trim switches, and the sticks I'm using are an early pair from the X9D that have the amplifiers. In passing, these give me a problem on openTx as throttle and elevator operated in reverse. Because of the amplifiers, I can't reverse them by changing the wiring, and there appears to be no option to reverse them in openTx (I have such an option in ersky9x).

I think the only hardware changes I have done to this board are to add 0.2uF on the soft power circuit, to hold soft power on during a watchdog reset, and to add 1100uF across the 3.3V power rails. I've soldered my big caps on the P8 connector for ease of connection. I don't currently have a speaker plugged in either, but didn't before when the power decayed over 300mS.

Just re-doing the test with your build2, and I'm back to a 300mS decay time!!!
However the image above is still valid, the reset pulse occurs at the same time, just after a small drop in the supply voltage, then no further resets throughout the power decay.

I'd really like to know why your reset during power off is so different to mine.

MikeBland, 2015-08-29:
Just to confirm, in case of differences, the processor on my board is a STM32F407ZET6, and I do have pin 143 lifted and tied to pin 144 as a hardware change.
To test the firmware, I used Zadig so I flashed both the bootloader and the firmware every time.
OPTCR shows AAED.
Just tested a watchdog reset (build2), and all worked correctly.

projectkk2glider, 2015-08-29:
Mike your results are strange. How do you explain not getting consistent VCC decay times with the same build? 

I am also really surprised that build 1 does not cause the power loop. Maybe you are missing it, the power loop on my board is only audible. I can hear the switching power supply turning on/off. My board is special that way, for example I sometimes get a strong audio noise from the switcher when the radio is powered by the USB only. The LCD and back-light are off during power loop, for all purposes the radio looks turned off, but it is not if you take a scope to it. It pulses on/of at about 2.7Hz.

My board is/has:
* second board received from FrSky
* basically stock with two modifications:
  * 1000uF VCC capacitor
  * lifted pin 143 and tied to 144
* without RTC battery
* only left gimbal connected, but that does not affect build 1
* charging LED added (should not affect this)
* same CPU  STM32F407ZET6
* no speaker
* no switches
* SD card plugged in
* default option bytes
* **does not have** added cap to power FET

projectkk2glider, 2015-08-29:
More tests with **build 1** using different VCC caps:
* 1000uF low ESR:  power loop
* 2 * 470uF (all other caps are not low ESR): loop
* 470uF + 330uF: loop
* 2 * 330uF: border case, sometimes without loop, other times one to three loops then power off
* 470uF: good (no loop)
* 330uF: good

Can you try other cap values on your board and watch for the power loop on scope.

MikeBland, 2015-08-29:
Nailed the voltage decay difference, it depends on the haptic setting. When I have it as quiet, with minimum length and strength (because it was irritating me), I get the long decay time, when it is set to the defaults I get the short decay time.

Testing with build 1 and a 1100uF, I still get no power loop. I get the same sort of output as I show on the 'scope trace above. The voltage decay time, to around 0.4V, is 500mS. I have the 'scope on 500mS/division, and the power just stays off even after 3.5 seconds from when it first start dropping.

I have 0.2uF between the junction of R71 and R73 and ground. These were needed to hold power on during a watchdog reset, before we turned the soft power back on in the bootloader.

When powering off, I see the supply voltage drop from 3.3 to 3.0 volts over 4mS, then the short NRST pulse occurs after which the supply voltage continues to decay, byt at a much slower rate.
I interpret this as the soft power is switched off, starting the voltage decay, then the processor enters standby, but takes 4mS to do so. During this time the processor is taking a 'high' current. On entering standby, we see the NRST pulse, and the processor current drops to almost nothing, so reducing the voltage decay rate.

I don't think I've ever seen the power loop occur.

What I don't understand is why your voltage decays to about 1.8V, then you get a long NRST pulse, while I don't.

projectkk2glider, 2015-08-29:
At least one mystery explained. I forgot to write above that I don't have the haptic motor connected at all. 

Perhaps we should stop investigating this, the way the code is written now should be much more bulletproof. 

One more question for you. Do you get an audible noise when the board is powered by the USB? Because I get it usually. Maybe my board is a bit different from yours.

kilrah, 2015-08-29:
For the record I also occasionally get the noise when powered by USB, can't find a definitive pattern. It also happened to me on the original prototype Taranis board (non-plus) so besides being annoyed by it I didn't give that much importance to it.

MikeBland, 2015-08-30:
Yes I get audible noise when powered by the USB. I think it is one or more of the switch mode regulators doing something when they see power applied on their output side and the inductors are then making the noise.

I agree the code should be OK, but it is very odd how we get very different responses of the NRST signal on power off.

commit 6d0d0f9b (2015-08-28): Merge pull request #2801 from opentx/projectkk2glider/issue_2790_power_loop

Fixes #2790: power on/off lop if big VCC capcitor was used
commit e4f7fd91 (2015-08-28): Fixes #2790: power on/off lop if big VCC capcitor was used