opentx/opentx #2792: RE; screen reader function for OPET TX ?.

bobaustralia, 2015-08-28:
HI, OPET TX. Not sure if this is an issue or not what I am asking if it is possible is to have it access accessibility software inserted in future versions of  OPET TX access accessibility is a free program produced by Microsoft for screen readers. I suffer from dyslexia and I use a screen reader program called read and write by text help on my computer. I have been flying model planes was several years and have used a spectrum DX7 radio but I want now upgrade to Tirana’s and if this program was in OPET TX will allow me to use the program by myself instead of having to asked for help other club members.
Bob.

projectkk2glider, 2015-08-28:
I always thought that the accessibility was provided by the operating system itself. Or does each program (in our case OpenTX Companion) have to have a special support for this kind of features?

bobaustralia, 2015-08-28:
Hi. The active accessibility has to be embedded in the software itself for the screen reader program to work US software is required by federal law to have it however most European software doesn’t have though I have an OCR program that is made by a Russian company it has active accessibility installed in it I hope this is helped.

 

Regards, Bob.

 

No virus found in this message.
Checked by AVG - www.avg.com
Version: 2015.0.6086 / Virus Database: 4409/10527 - Release Date: 08/28/15

bsongis, 2015-08-28:
I was not aware of that, I will see if Qt does provide an easy way to bring
to Companion the accessibility feature

bobaustralia, 2015-08-29:
Hi. One other thing I meant to mention to the forum was that open pilot software has active accessibility built-in as downloaded it and mucking around with it before wanting to buy a quad with  CC3D may be?.

 

Regards, Bob.

 

PS. Anybody’s wondering these posts are dictated using dragon 12.

bsongis, 2015-08-29:
I did some commits in OP. Good to know accessibility is implemented there,
they also use Qt!

closed 2017-05-11 by mpaperno